OpenIntelWireless/itlwm #645: Can't Load itlwm kext with kextutil after loading to /tmp

divrse, 2021-07-12:
I've used airportitlwm on big sur with an intel ax201 and it works alright but disconnects / have to disable and re enable wifi randomly about every hour.
I thought i'd give itlwm + heli port a try and put it in my kexts folder along with config.json

upon it not working, i followed the steps on your site to manually load.
```bash
cp -R itlwm.kext /tmp
sudo chown -R root:wheel /tmp/*.kext
sudo kextutil -v /tmp/*.kext
```
unfortunately, my mac required a reboot for this kext to be loaded.
Now, here is the problem.
I used /tmp/itlwm.kext as the location, and tmp gets cleared every restart.
I even tried to copy the folder using /etc/rc.local but it didn't work.

Now my mac is trying to load from tmp and I cannot unload because it was never loaded in the first place.

I cannot add the kext again from a different location it fails (assuming because it has the same name (com.zxystd.itlwm)

and I cannot clear the tmp kext, clearing it with kextunload and kmutil do absolutely nothing.

So the only way to resolve this is change the directory and or find a way to permanently clear the kext. both of which I haven't found a way to do so.

zxystd, 2021-07-12:
Why not put it in bootloader?

closed 2021-07-12 by zxystd

divrse, 2021-07-13:
I've put in in bootloader and it fails to load.

williambj1, 2021-07-13:
Have you tried to clean/rebuild the kernel cache and what are the terminal outputs? Logs from the bootloader might be useful as well.