OpenIntelWireless/itlwm #646: n7265 wifi on-off looping and disconnecting

edwardwich, 2021-07-12:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment

- Kext Version: v2.0.0-alpha/v1.3.0 stable
- WiFi Card Model:  Intel Wireless-N 7265 Wireless Network Adapter
- PCI Product ID: 0x8086  0x95A
- macOS Version: 11.4 (20F71)

Description

It is difficult to connect to the net at long distances (this problem does not exist in Linux and Windows)

Bug Report Archive
[Log_2021-07-13_04-11-31.log.zip](https://github.com/OpenIntelWireless/itlwm/files/6804923/Log_2021-07-13_04-11-31.log.zip)

Kext Download Source
https://github.com/OpenIntelWireless/itlwm/releases

zxystd, 2021-07-13:
Works find with 1.2.0?

edwardwich, 2021-07-13:
im test 1.2.0 not work never connect to my modem

edwardwich, 2021-07-13:
but v2.0.0-alpha work better

edwardwich, 2021-07-13:
long distance = It is not connected to the modem from a distance of more than 2 meters

edwardwich, 2021-07-16:
i tested
The results are the same

williambj1, 2021-10-02:
Log files are not properly collected, can't see any evidence of disconnection. Will consider reopening once properly generated logs from v2.0.0 and v2.1.0 are provided.

closed 2021-10-02 by williambj1