OpenIntelWireless/itlwm #647: Wireless adapter does not turn on after sleep

startergo, 2021-07-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Itlwm 1.3.0
- WiFi Card Model: Centrino Advanced-N 6205 [Taylor Peak] 
- PCI Product ID: DID 0082 SVID 8086  SDID 1301
- macOS Version: 10.13.6

**Description**
Wireless Interface would not turn on after sleep. Need to reboot the system to get it working.

**Bug Report Archive**
[bugreport_39195.zip](https://github.com/OpenIntelWireless/itlwm/files/6834748/bugreport_39195.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.3.0

zxystd, 2021-07-17:
@startergo The problem is fixed in alpha-v2.0.0, can you try it?

startergo, 2021-07-17:
Works now great!. I was trying to build the latest commit from source, but I got some errors in Xcode:
[image]

closed 2021-07-17 by startergo