OpenIntelWireless/itlwm #648: Dependency com.apple.iokit.IO80211Family was not found for kext com.zxystd.AirportItlwm

startergo, 2021-07-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportitlwmHighSierra v2.0.0 alpha
- WiFi Card Model: Centrino Advanced-N 6205 [Taylor Peak]
- PCI Product ID: DID 0082 SVID 8086 SDID 1301
- macOS Version: 10.13.6

**Description**
Kext does not load due to a missing dependency:
```
Dependency com.apple.iokit.IO80211Family was not found for kext com.zxystd.AirportItlwm
```

```
kextstat | grep com.apple.iokit.
   11   32 0xffffff7f80f64000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) 95DA39BB-7C39-3742-A2E5-86C555E21D67 <7 6 4 3>
   12   39 0xffffff7f80a94000 0x35000    0x35000    com.apple.iokit.IOPCIFamily (2.9) F763C275-306F-358F-B6B0-0FD93486BD5F <7 6 5 4 3>
   15    9 0xffffff7f81120000 0x30000    0x30000    com.apple.iokit.IONetworkingFamily (3.4) 5A99D802-B477-3D83-9F76-304605479B9F <7 6 5 4 3 1>
   16    2 0xffffff7f81674000 0x2d000    0x2d000    com.apple.iokit.IOTimeSyncFamily (680.15) 6DD2AAD0-FC4B-32EC-B0FD-AC55A727A789 <15 5 4 3 1>
   17    5 0xffffff7f8184b000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) D2F2FBDF-4EE4-38BA-99F5-B699F886F413 <5 4 3>
   18    9 0xffffff7f80e5d000 0x28000    0x28000    com.apple.iokit.IOStorageFamily (2.1) F27A8A2A-6662-3608-83BD-415037509E01 <7 6 5 4 3 1>
   24   19 0xffffff7f80f9c000 0x79000    0x79000    com.apple.iokit.IOUSBHostFamily (1.2) 93B5E6D7-8F74-39E2-A9C6-6B25F334733D <23 22 7 6 5 4 3 1>
   27    3 0xffffff7f80a4b000 0x2b000    0x2b000    com.apple.iokit.IOSCSIArchitectureModelFamily (404.30.3) 2E28EF6C-9898-3506-AB5A-433B83CDD983 <6 5 4 3 1>
   28    1 0xffffff7f8103d000 0x19000    0x19000    com.apple.iokit.IOSCSIBlockCommandsDevice (404.30.3) 66A63094-72B6-3E25-BE1C-6CEB28186627 <27 18 6 5 4 3 1>
   29    1 0xffffff7f81163000 0x2e000    0x2e000    com.apple.iokit.IOUSBMassStorageDriver (140.70.2) 91925DA5-1D2A-3444-A53B-1432B6363BA7 <27 24 18 6 5 4 3 1>
   55    6 0xffffff7f81259000 0x7b000    0x7b000    com.apple.iokit.IOHIDFamily (2.0.0) 0DD0A3A0-A812-3B96-AEEB-6F257C2147CA <7 6 5 4 3 2 1>
   58    2 0xffffff7f817ce000 0x4000     0x4000     com.apple.iokit.IOSMBusFamily (1.1) 9CFF73D8-2872-3FC0-AEE5-334587EAA4D0 <5 4 3>
   61    1 0xffffff7f81198000 0x9b000    0x9b000    com.apple.iokit.IOUSBFamily (900.4.1) 2837DD62-F2B0-3357-A164-D44FEF6810E0 <24 23 12 7 5 4 3 1>
   67    2 0xffffff7f81cdc000 0x1a000    0x1a000    com.apple.iokit.IOAHCIFamily (288) 50673EBC-47A0-329A-B58C-081A6EDF6BF4 <5 4 3 1>
   69    2 0xffffff7f81060000 0x73000    0x73000    com.apple.iokit.IOFireWireFamily (4.7.2) 35741F64-7484-39D5-81B9-9240CD5A50DB <5 4 3 1>
   73    0 0xffffff7f81d06000 0x27000    0x27000    com.apple.iokit.IOAHCIBlockStorage (301.40.2) 0D9325A5-E989-39F7-B3CF-75FC51C60627 <67 18 6 5 4 3 1>
   88    3 0xffffff7f81a7e000 0xf2000    0xf2000    com.apple.iokit.IOBluetoothFamily (6.0.7f22) FF1CAA92-5B92-3233-B7F0-FC70FFE348E8 <24 17 11 7 6 5 4 3 1>
   89    2 0xffffff7f81b84000 0xb000     0xb000     com.apple.iokit.IOBluetoothHostControllerTransport (6.0.7f22) D1E0D9F2-51AD-3F29-A376-76C32B87A97B <88 24 12 11 7 6 5 4 3 1>
   90    1 0xffffff7f81b9a000 0x20000    0x20000    com.apple.iokit.IOBluetoothHostControllerUSBTransport (6.0.7f22) 4CB2D35A-475F-344F-A977-96EDC491651E <89 88 24 12 11 7 6 5 4 3 1>
   91    0 0xffffff7f81bba000 0xc000     0xc000     com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport (6.0.7f22) 76A5A31F-D846-3945-92AB-9413637E938D <90 89 88 24 12 11 7 6 5 4 3>
   95   16 0xffffff7f8193c000 0x49000    0x49000    com.apple.iokit.IOGraphicsFamily (519.23) 81D774E8-DD11-3866-B90A-E855215EDB60 <12 7 5 4 3 1>
   97    2 0xffffff7f817b3000 0x19000    0x19000    com.apple.iokit.IOSurface (211.15) 37BCC686-AF99-3381-AC4F-E54A347E0F40 <7 6 5 4 3 1>
   98    2 0xffffff7f8233e000 0x99000    0x99000    com.apple.iokit.IOAcceleratorFamily2 (378.28) D342FE41-0457-3586-ABB0-38E8E3947C4F <97 95 30 12 7 6 5 4 3 1>
  101    1 0xffffff7f817d2000 0x5000     0x5000     com.apple.iokit.IOSlowAdaptiveClockingFamily (1.0.0) CBE56568-AC91-3F56-A06E-9A29D6F70BDA <7 6 5 4 3 1>
  108    0 0xffffff7f819fb000 0xe000     0xe000     com.apple.iokit.IOFireWireIP (2.2.9) AD37305B-E664-3DFB-961B-FB9260E508E9 <69 15 5 4 3 1>
  109    4 0xffffff7f81992000 0x10000    0x10000    com.apple.iokit.IONDRVSupport (519.21) B9EB0174-55E4-3CB1-9802-E74BB1BD5EF0 <95 12 7 5 4 3>
  118    2 0xffffff7f817e6000 0xe000     0xe000     com.apple.iokit.IOSerialFamily (11) 6FE74518-ECD9-32D6-88BE-8F5BCAC911D4 <7 6 5 4 3 1>
  125    4 0xffffff7f815cd000 0x3f000    0x3f000    com.apple.iokit.IOAudioFamily (206.5) 2F090399-C028-3F14-AE4F-BF49A3D602E5 <124 5 4 3 1>
  126    2 0xffffff7f8263c000 0xc000     0xc000     com.apple.iokit.IOHDAFamily (281.52) 3D7ADFBB-A913-31F3-A713-32A20B7B2249 <5 4 3 1>
  130    2 0xffffff7f81bc6000 0x7000     0x7000     com.apple.iokit.IOEthernetAVBController (1.1.0) 0B1D1184-C655-3A04-A741-4704652016B6 <15 5 4 3 1>
  132    0 0xffffff7f81c78000 0x10000    0x10000    com.apple.iokit.IOAVBFamily (683.1) 727F5C95-C893-3F15-9072-A409247A0DFF <131 130 16 15 6 5 4 3 1>
  133    0 0xffffff7f81a6e000 0xe000     0xe000     com.apple.iokit.IOBluetoothSerialManager (6.0.7f22) B5282E7B-2280-3FF0-BC90-6C1A3E31EDBC <118 7 5 4 3 1>
  134    0 0xffffff7f81157000 0xa000     0xa000     com.apple.iokit.IOUserEthernet (1.0.1) 227C4C3C-7AA7-366C-A2FA-73609D667F8F <15 6 5 4 3 1>

```

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases

williambj1, 2021-07-17:
IO80211Family does not load by itself prior to macOS 11. Use OpenCore to force load whatever is missing. Please read the docs.

closed 2021-07-17 by williambj1

startergo, 2021-07-17:
Thanks. It would be nice to expand on it in the manual. For the benefit of the users:

```
</dict>
		<key>Force</key>
		<array>
			<dict>
				<key>Arch</key>
				<string>Any</string>
				<key>BundlePath</key>
				<string>System/Library/Extensions/corecapture.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/corecapture</string>
				<key>Identifier</key>
				<string>com.apple.driver.corecapture</string>
				<key>MaxKernel</key>
				<string>17.07.00</string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
			<dict>
				<key>Arch</key>
				<string>Any</string>
				<key>BundlePath</key>
				<string>System/Library/Extensions/IO80211Family.kext</string>
				<key>Comment</key>
				<string></string>
				<key>Enabled</key>
				<true/>
				<key>ExecutablePath</key>
				<string>Contents/MacOS/IO80211Family</string>
				<key>Identifier</key>
				<string>com.apple.iokit.IO80211Family</string>
				<key>MaxKernel</key>
				<string>17.07.00</string>
				<key>MinKernel</key>
				<string></string>
				<key>PlistPath</key>
				<string>Contents/Info.plist</string>
			</dict>
```