OpenIntelWireless/itlwm #651: can't connect wap2 802.11

twogoods, 2021-07-20:
**Have You Read Our Docs**
Yes 

**Are You Reporting A Bug**
maybe

**Environment**
- Kext Version:  1.3.0 
- WiFi Card Model:  3160 ac
- PCI Product ID: 
- macOS Version:  10.15.6

**Description**
can't connect wap2 802.11

**Bug Report Archive**
[Report_2021-07-20_20-17-44.log](https://github.com/OpenIntelWireless/itlwm/files/6848203/Report_2021-07-20_20-17-44.log)

zxystd, 2021-07-21:
The log is incomplete, please collect again.

- Add msgbuf=1048576 into boot-args using either nvram or your bootloader's config

- Run the following command in terminal:
```
sudo dmesg | grep -i "itlwm" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log
```
A log file will then be created on your Desktop.

twogoods, 2021-07-21:
@zxystd 
[Log_2021-07-21_19-11-43.log](https://github.com/OpenIntelWireless/itlwm/files/6855133/Log_2021-07-21_19-11-43.log)

thanks~

zxystd, 2021-07-23:
Can't find what's wrong there, seems that no EAPOL frames coming, I need more information. BTW, is it an enterprise network? what's the encryption algorithm of it?
```
sudo dmesg | grep -E "itlwm|Airport|EAPOL" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log
```

danielstuart14, 2021-07-23:
Still the same, but I found out that I'm able to connect to one of my 5G WiFis (VIVOFIBRA-9FD8-5G2), but not to my other 5G (VIVOFIBRA-9FD8-5G) (lower channel) and 2.4Ghz WiFi. Here is the log from the latest commit:
[Log_2021-07-23_12-18-59.log](https://github.com/OpenIntelWireless/itlwm/files/6870023/Log_2021-07-23_12-18-59.log)

zxystd, 2021-07-24:
@danielstuart14 
```
[  443.299922]: itlwm: iwm_run
[  443.299931]: itlwm: iwm_phy_ctxt_cmd
[  443.300024]: itlwm: ieee80211_recv_disassoc
[  443.300033]: itlwm: Disassoc received, reason 7
[  443.300040]: itlwm: ieee80211_recv_disassoc 3197 nstate=3
```
Your router refuse the connection, can you check the router setting?

twogoods, 2021-07-24:
> Can't find what's wrong there, seems that no EAPOL frames coming, I need more information. BTW, is it an enterprise network? what's the encryption algorithm of it?
> 
> ```
> sudo dmesg | grep -E "itlwm|Airport|IO80211|EAPOL" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log
> ```

[Log_2021-07-24_15-17-59.log](https://github.com/OpenIntelWireless/itlwm/files/6872063/Log_2021-07-24_15-17-59.log)

enterprise network,but i don't know what algorithm of it😓

danielstuart14, 2021-07-26:
I don't know if it'll help, but I found out that the issue is when the WiFi has a repeater. Maybe it's trying to connect to one of the routers and changes to the other in the middle of the process of authentication. I can guarantee that the WiFi works as I use it both on my phone and on my pc running Windows. I'll try to debug it a little more later.

williambj1, 2021-07-28:
[802.1X_Verbose_Logging_Instructions.pdf](https://github.com/OpenIntelWireless/itlwm/files/6892756/802.1X_Verbose_Logging_Instructions.pdf)

@twogoods Could you follow this guide and attach the output? Thanks!

zxystd, 2021-07-30:
@twogoods  I am finding why macOS 10.15 can't recognize the WPA/WPA2 Enterprise encryption, for the temporary solution, you can manually add your WiFi info in the network setting, and then connect.
[image]

twogoods, 2021-07-31:
> @twogoods I am finding why macOS 10.15 can't recognize the WPA/WPA2 Enterprise encryption, for the temporary solution, you can manually add your WiFi info in the network setting, and then connect.
> [image]

Work !  thanks ~

closed 2021-07-31 by twogoods