OpenIntelWireless/itlwm #652: Wifi connects and soon get disconnected

ultinux, 2021-07-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm v2.0.0 alpha
- WiFi Card Model: ac9462
- PCI Product ID: Unknown
- macOS Version: Big Sur 11.5

**Description**
It connects to the wifi (5Ghz, WPA2 PSK), internet works, but gets disconnected after few seconds. Works well with AirportItlwm v1.2.0 except getting disconnected randomly within a long term. v1.3.0 didn't work.

**Bug Report Archive**
[Log_2021-07-23_00-31-53.log](https://github.com/OpenIntelWireless/itlwm/files/6863506/Log_2021-07-23_00-31-53.log)

**Kext Download Source**
From Github Release page

ultinux, 2021-07-23:
Just tried to connect my Galaxy S9+ 5Ghz hotspot, and it works flawlessly with v2.0.0 alpha.
Here is the log when I connected to the hotspot. Still disconnects when I connect to my home router.
[Log_2021-07-23_22-19-36.log](https://github.com/OpenIntelWireless/itlwm/files/6869023/Log_2021-07-23_22-19-36.log)

ultinux, 2021-07-23:
Fixed! https://github.com/OpenIntelWireless/itlwm/issues/652

closed 2021-07-23 by ultinux

williambj1, 2021-07-23:
Duplicate of #628