OpenIntelWireless/itlwm #656: Low and unstable speeds in 2.0.0 release

Mahas1, 2021-08-08:
**Have You Read Our Docs**
- Yes

**Are You Reporting A Bug**
 - Yes

**Environment**

- Kext Version: 2.0.0 release
- WiFi Card Model: AC9560
- PCI Product ID:  ven: 8086, dev: A370
- macOS Version: 11.5.1

**Description**
Slow speeds in 2.0.0 release
I get 90mbps (same as windows) in 1.3.0 release, while 20-25mbps in 2.0.0 release.

**Bug Report Archive**

Heliport logs (Used Airportitlwm and used heliport for logs):
[heliport_log_-_bugreport_46247.zip](https://github.com/OpenIntelWireless/itlwm/files/6950367/heliport_log_-_bugreport_46247.zip)

Terminal logs (Airportitlwm):
[airportitlwmlog_-_Log_2021-08-08_12-40-00.log](https://github.com/OpenIntelWireless/itlwm/files/6950368/airportitlwmlog_-_Log_2021-08-08_12-40-00.log)

**Kext Download Source**
 - releases page of this repo

yusufgngor, 2021-08-09:
I got same issue as well. My speed decreased almost 75% and while speedtesting the neddle is  making instant movements. 

- Kext: Airport 2.0.0 Bigsur 
- WiFi Card Model: AC9560
- macOS Version: 11.5.1

Dave1482, 2021-08-10:
I have tried AirPortItlwm v2.1.0-alpha with the AX210, speeds are around 5Mbps for me. However, itlwm v2.1.0-alpha + HeliPort gives me about 450Mbps down and 23Mbps up (same as Windows)! No Airdrop, etc. unfortunately, but what can I do about it? lol I'm quite happy with the progress!

wydy, 2021-08-10:
Same problem on AC9560 while using AirPortItlwm.
But only 5ghz networks affected, 2.4ghz is significant improved compared to 1.3.0

schmosbyy, 2021-08-10:
Same problem with AC8265 , speed becomes 0.5 mpbs for 5ghz on v2 release, while on v1.3.0 the speed is around 91mpbs.
-Using v2.0.0 + heliport.
Have tried 1.3 to 1.4.1 versions of heliport, same results.
Seems to be some issue with v2.0.0 of itlwm.

For now have reverted to v 1.3.0 for stability

zxystd, 2021-08-10:
[itlwm_disable_11ac.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6962365/itlwm_disable_11ac.kext.zip)
what about the disabled 11ac version?
please collect the running logs if you want to issue something, that may help me to trace the problem. here is the guide: https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

wydy, 2021-08-10:
> [itlwm_disable_11ac.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6962365/itlwm_disable_11ac.kext.zip)
> what about the disabled 11ac version?
> please collect the running logs if you want [...]

New version seem to be work, thanks!

Mahas1, 2021-08-11:
@zxystd Here are the results with the modified kext you sent
2.4ghz - 25mbps down, 50mbps up
5ghz - 50mbps down, 5mbps up

heliport logs
[bugreport_29636.zip](https://github.com/OpenIntelWireless/itlwm/files/6965773/bugreport_29636.zip)

for comparison, both 2.4ghz and 5ghz gave me 90 up and 90 down in 1.3.0

DianaSensei, 2021-08-12:
I have same problem with v2.0 stable and v2.1-alpha
In v1.3: 90mbps down, 75mbps up
In v2.0: 40mbps down, 28mbps up
In v2.1-alpha: 20mbps down, 19mbps up

Wifi Card Model: Intel AC 9560
MacOS: 11.5.1

JulioCMLP, 2021-08-13:
I got same problem but when my bluetooth is connected to any devices 
Kext: Airport 2.0.0 Bigsur
WiFi Card Model: AC9462
macOS Version: 11.5.2

alinach120602, 2021-08-13:
Getting excellent WiFi AC 80 MHz 867/867 and real World 640Mbit/s Down and 380Mbit/s Up, measured with iPerf.
zxystd you did an amazing Job with this and I must congratulate and thank you for your Hard work!

Kext: AirportItlwm 2.0.0 Release
WiFi Card Model: Intel AX200
macOS Version: Big Sur 11.5.2

Thank you very much.

xawlw, 2021-08-14:
I have the same problem with the 5Ghz network, I'm getting 2Mbps up and down although I have a 100mbps connection, However, this issue happens on 1.3 and 1.2 as well

Card: AX200
macOS: 11.4

xawlw, 2021-08-14:
> I have the same problem with the 5Ghz network, I'm getting 2Mbps up and down although I have a 100mbps connection, However, this issue happens on 1.3 and 1.2 as well
> 
> Card: AX200
> macOS: 11.4

I forgot to mention that 2.4Ghz is working perfectly and I'm getting full 100mbps on 2.0 and the prior versions

williambj1, 2021-10-02:
The latest alpha paired with the `-novht` boot-args flag can be used as a workaround.

Router compatibility issues are extremely hard to debug without the hardware on hand. Pull requests are always appreciated.

Mahas1, 2021-10-02:
> The latest alpha paired with the `-novht` boot-args flag can be used as a workaround.
> 
> Router compatibility issues are extremely hard to debug without the hardware on hand. Pull requests are always appreciated.

Hi, thank you for the reply. This does seem to work, but the speeds are not as high as what 1.3.0 gives. I will use this for now, thanks again.

HaCkEuR0ElItE, 2021-11-24:
Hello, I have the same problem with my EliteBook 850-G3 and Wireless-AC 8260, good speed with my old router (AC) 100mbps on down with the 2.0.0 and 2.1.0 version but with my new router (AX), I have a very low speed (0.5mbps) on down with 2.0 and 2.1 version but I have better speed (60/70mbps) with the 1.3 version on 802.11n 5GHz, 40 MHZ. Is not a problem that the wireless card is AC and the router is AX ? If you need, I can do a log on 2.0 / 2.1 and 1.3 ?

Stevemoretz, 2021-11-24:
> Hello, I have the same problem with my EliteBook 850-G3 and Wireless-AC 8260, good speed with my old router (AC) 100mbps on down with the 2.0.0 and 2.1.0 version but with my new router (AX), I have [...]

Weird maybe new router is slower too

williambj1, 2021-11-24:
@HaCkEuR0ElItE @Stevemoretz 

Might be related to WPA3.

https://github.com/OpenIntelWireless/itlwm/issues/701#issuecomment-969014672

Stevemoretz, 2021-11-24:
> @HaCkEuR0ElItE @Stevemoretz
> 
> Might be related to WPA3.
> 
> [#701 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/701#issuecomment-969014672)

Thank you but it has WPA 2 not 3

williambj1, 2021-11-24:
@Stevemoretz Maybe consider trying that version regardless? If the issue still exists logs will certainly be helpful.

Stevemoretz, 2021-11-24:
Great idea, but unfortunately my new other is somewhere else which I go for vacations so I can't test it pretty soon.

@HaCkEuR0ElItE you might wanna give it a try, just use itwlm instead of the airport version. (it has a separate program for its ui which you need to install as well)

HaCkEuR0ElItE, 2021-11-24:
> > @HaCkEuR0ElItE @Stevemoretz
> > Might be related to WPA3.
> > [#701 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/701#issuecomment-969014672)
> 
> Thank you but it has WPA 2 not 3

Yes it's also WPA 2 (Personal) and not WPA3
It seems to me to have noticed and it's quite weird that I had 0.5mbps on the whole system but when I use a VPN like for example on the Opera browser, the connection is faster to load website than without VPN, I did not test speed in relation to that, I will check all this this weekend as well as do the logs to see you what is wrong and check with the itwlm version :)

HaCkEuR0ElItE, 2021-12-10:
You can check the manual log for 2.1.0 here:  
[Log_2021-12-10_15-36-15.log](https://github.com/OpenIntelWireless/itlwm/files/7693541/Log_2021-12-10_15-36-15.log)
I have also the apple wireless log file, but I can't post that here (refused by github) 
You can also see 4 speedtest (I don't know why the app and the website of speedtest.net have the better and the good speed but it's not the real speed for the loading of pages (On Safari, Opera, Firefox same things), the google speedtest is the good speed loading and it's very slow).
[image]
[image]
[image]
[image]

HaCkEuR0ElItE, 2021-12-13:
I think maybe I know what's wrong with it, but I don't know if you have the same setup as me.
In my living room, I actually pick up my 3 access points (main modem, AP Repeater in the upstairs bedroom and AP in my living room). The AP Repeater is upstairs just above my living room so I am picking up duplicate 2.4 and 5GHz ssids with very good dBm signals and I don't know if these would not be involved in a "conflict". But I also pick up the SSID of my kitchen much more weakly in my living room.
What is he making me say this? The 3 APs are the same WIFI models and the same brand (A whole SagemCom system in WIFI6 AX). In my living room I have a page load rate of 0.33mbps, but the speed issue seems to be download only and not upload. On the other hand, if I place myself at the end of my network, I only receive the wifi in my kitchen and the page loading speed is quite good 26.7mbps, at least, all the pages are loaded instantly.
In my living room, I have to connect through a VPN to have a fast page load speed. 

This is the Wifi scan in my living room:
[image]
This is the wifi speed in my living room:
[image]
This is the wifi scan at the end of my network:
[image]
This is the wifi speed at the end of my network:
[image]
Apple Wifi info at the end of my network: 
[image]
Apple wifi info in my living room:
[image]

I hope that it can help you to find the problem @zxystd.

williambj1, 2022-03-03:
Should be fixed in v2.2.0. Feel free to add more comments if that is not the case.

closed 2022-03-03 by williambj1

Mahas1, 2022-03-12:
Speeds are back to normal when used with the `-novht` boot arg, thanks a lot!

zxystd, 2022-03-12:
what about using v2.1.0 or v2.2.0-alpha version?

Mahas1, 2022-03-12:
> what about using v2.1.0 or v2.2.0-alpha version?

@zxystd 2.1.0 gives low speeds, almost the same as 2.0.0, for which the issue was opened. Speeds are back to normal with the latest 2.2.0 Alpha, which the `-novht` boot arg. Without the `-novht` boot arg, it gives low speeds again.

HaCkEuR0ElItE, 2022-03-13:
Hi,
It's the same for me, again low download speed with the 2.1 and  2.2 with AC activated on a AX router.

[image]
[image]

husnainalam888, 2024-09-14:
Issue with AirportTwlm Kext: Slow Internet Speeds

Hi,

I've been using the AirportTwlm kext on my Intel-based Hackintosh for the WiFi card, but I've noticed that the internet speeds are significantly slower compared to what I expect. After troubleshooting, I switched to using itwlm with HeliPort, and the performance has improved dramatically—it works perfectly for me now.

I would recommend others facing similar speed issues to try this combination. Perhaps there could be an improvement or alternative setup with AirportTwlm that addresses this speed issue?

Thanks for all the hard work on this project!

Best,
Husnain AlamIssue with AirportTwlm Kext: Slow Internet Speeds

Hi,

I've been using the AirportTwlm kext on my Intel-based Hackintosh for the WiFi card, but I've noticed that the internet speeds are significantly slower compared to what I expect. After troubleshooting, I switched to using itwlm with HeliPort, and the performance has improved dramatically—it works perfectly for me now.

I would recommend others facing similar speed issues to try this combination. Perhaps there could be an improvement or alternative setup with AirportTwlm that addresses this speed issue?

Thanks for all the hard work on this project!

Best,
Husnain Alam