OpenIntelWireless/itlwm #659: AirportItlwm

calibra43, 2021-08-10:
**Environment**

- Kext Version: 2,0
- WiFi Card Model: 9260ngw
- PCI Product ID: 8086,2526
- macOS Version: 10.15.7

HI ALL. need help
 intel ac9260ngw not find whith my mac os  catalina
pc lenovo thinkcentre m720q b360chipset
hackintool find pci wifi

Wireless-AC9260 Network Controller pci 8086,2526  PciRoot(0x0)/Pci(0x1C,0x0)/Pci(0x0,0x0)

I am install Airportitlwm2,0 for catalina itlwm2,0. my system not found this wifi,bluethooth worked fine!!!
i see whith terminal kext loaded fine. Heliport write itlwm not load.i see log heliport,its write itlwm.kext not load.
maybe this kext not worked whith my intel 9260?

Bootloader Opencore 7.0

Thanks and sorry for my english

jackjack1-su, 2021-08-11:
Never use airportitlwm and itlwm together.
If you use Airportitlwm, you no longer need Heliport. Airportitlwm supports the original macOS Wifi Interface.(System Preferences->Network->WiFi)
Or use itlwm.kext with Heliport.

calibra43, 2021-08-11:
I used Airportitlwm without itlwm.Heliport

williambj1, 2021-08-11:
It's necessary to force load IO80211Family in macOS 10.15, please read the docs for more info.

closed 2021-08-11 by williambj1