OpenIntelWireless/itlwm #660: Connection With Stock Android Phones

singhalrishi27, 2021-08-11:
**Environment**
- Kext Version: 2.1.0/2.0.0
- WiFi Card Model: Wireless AC 7260
- PCI Product ID: 0x8086. 0x4470
- macOS Version: 12 Monterey
**Description**
When I Use Xiaomi Vivo or any other Phone as Hotspot Speeds are generally stable and Hotspot connects Instantly But
when i use any Stock android Based phone wifi sometimes connects sometimes it doesn't.
Two Possibilities
1. Wifi Connected Successfully
Internet does not work even after successful connection but when i use my phone's USB Tethering to connect internet to
my laptop then Connect to Wifi via my phone, Internet Works Even after disconnecting USB Tethering Inshort to get
internet Access via Pixel i will have to be connected with any other network. Even then Speeds are Slow AF My phone is
giving me 43 mbps of speeds and on my laptop connected via wifi is giving me 2 to 1 mbps of speeds.
2.) Wifi Simply Does not connect
Returns a error Pixel 4a Can't be Joined
**Bug Report Archive**
I tried with Airportitlwm 2.0.0 and alpha 2.1.0
Even switched Operating System Big Sur and Monterey
Issue Still Remains
[bugreport_55278.zip](https://github.com/OpenIntelWireless/itlwm/files/6966557/bugreport_55278.zip)
[Kext Download Source - Release Section of your Reposiory](https://github.com/OpenIntelWireless/itlwm/releases)

williambj1, 2021-08-11:
Just a quick reply, I'm using LineageOS 16 and 18, working flawlessly for me. Will look into the logs later.

singhalrishi27, 2021-08-11:
> Just a quick reply, I'm using LineageOS 16 and 18, working flawlessly for me. Will look into the logs later.

Im using Pixel Experience, Evolution X also my phone works really well in windows using same laptop give me about 95% of speeds but with airport itlwm installed in macos it's shame

zxystd, 2021-08-16:
@singhalrishi27 Attached log is incomplete, please follow the guide to collect again.
https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#runtime-logs

singhalrishi27, 2021-08-20:
[Log_2021-08-20_08-20-17.log](https://github.com/OpenIntelWireless/itlwm/files/7018808/Log_2021-08-20_08-20-17.log)
[Log_2021-08-20_08-21-06.log](https://github.com/OpenIntelWireless/itlwm/files/7018809/Log_2021-08-20_08-21-06.log)
[image]

williambj1, 2021-08-20:
Hi, have you tried itlwm (not AirportItlwm) as well?

singhalrishi27, 2021-08-20:
> Hi, have you tried itlwm (not AirportItlwm) as well?

  
[image]
this is the result using heliport

singhalrishi27, 2021-08-20:
[image]
This is using same network using Horndis usb tethering

williambj1, 2021-08-20:
HI, could you please check if the phone is creating a 2.4GHz ***40MHz*** hotspot? Have you tried creating a 5GHz hotspot as well?

singhalrishi27, 2021-08-20:
Yup I tried 2.4Ghz and 5Ghz hotspot as well

singhalrishi27, 2021-08-20:
@williambj1 I even reinstalled Big Sur and Monterey 3 times Also reseted Network Preferences it doesn't seem to be an Operating System issue...

williambj1, 2021-10-02:
@singhalrishi27 Hi, do you have root access to the phone? I'm curious about the configuration file for hostapd, which can be found under `/data/vendor/wifi/hostapd`.

Here's mine

```
interface=wlan1
driver=nl80211
ctrl_interface=/data/vendor/wifi/hostapd/ctrl
ssid2=***
channel=0
acs_exclude_dfs=1
freqlist=2412-2462,5170-5865
ieee80211n=1
ieee80211ac=1

hw_mode=any
ht_capab=[HT40+]
vht_oper_chwidth=1
ignore_broadcast_ssid=0
wowlan_triggers=any
wpa=2
rsn_pairwise=CCMP
wpa_passphrase=***
```