OpenIntelWireless/itlwm #661: AC9260ngw not detected

calibra43, 2021-08-12:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 2.0
- WiFi Card Model: intel ac9260ngw
- PCI Product ID: pci 8086 2526
- macOS Version: 10.15.7 catalina and big sur 11.3

**Description**
HI. my pc not detected wifi card.i read log 
 itlwm: itlwm: failed to read nvm 
What solution for repaired?
Thanks
[Log_2021-08-12_19-07-55.log.zip](https://github.com/OpenIntelWireless/itlwm/files/6977399/Log_2021-08-12_19-07-55.log.zip)
[image]

williambj1, 2021-08-12:
Hi, could you please check if wifi works in Linux as well? Looks like the card is locked into airplane mode.

calibra43, 2021-08-13:
Ubuntu 21.04
Carte not detected,just detect Windows

williambj1, 2021-08-13:
> Ubuntu 21.04
> Carte not detected,just detect Windows

Unfortunately this is a hardware issue and we can't do anything from our side. Consider contacting the OEM for help.

closed 2021-08-13 by williambj1

calibra43, 2021-08-13:
Hardware intel wifi? or Pc?
Thank you

williambj1, 2021-08-13:
We are not sure either, maybe consider checking both.

calibra43, 2021-08-13:
i bought another card. Intelc ac3165.will install let you know
Thank you for support