OpenIntelWireless/itlwm #663: Building error

startergo, 2021-08-15:
May I suggest adding ```$(SRCROOT)/MacKernelSDK-master/Headers``` to the ```System Header Search Path``` in the ```Search Paths``` to enable successful build after adding ```MacKernelSDK``` to the project folder?
Without it it will not correctly build.

zxystd, 2021-08-16:
The SDK name is `MacKernelSDK`, not `MacKernelSDK-master`. User mistakes are out of our control and we will not considering about that.
Anyway, thank you for the advice.

closed 2021-08-16 by zxystd

startergo, 2021-08-16:
It depends. I download the ```MacKernelSDK-master``` from source so that is how it is labeled by default in my folder. It can be renamed of course. My point is that ```MacKernelSDK``` is not set by default in the search path and the user has to add it before the building process.

williambj1, 2021-08-16:
It has, otherwise it won't compile at all. Read the Xcode project file and the GitHub Actions yaml for details.

zxystd, 2021-08-16:
```
git clone https://github.com/OpenIntelWireless/itlwm.git
cd itlwm
git clone --depth=1 https://github.com/acidanthera/MacKernelSDK.git
```
This is the correct usage of GitHub project.