OpenIntelWireless/itlwm #665: Feature request: Open Wireless Link while using HeliPort

JJTech0130, 2021-08-18:
Just a thought, do you think it would be possible to integrate with https://github.com/seemoo-lab/owl when not using Airport? This would allow for some AirDrop functionality (with https://github.com/seemoo-lab/opendrop) and complete Open-Source. Obviously this might not work, what do I know, but I just wanted to share my idea.

JJTech0130, 2021-08-18:
Just saw https://github.com/OpenIntelWireless/HeliPort/issues/144. Can this be done completely in the user-space, without changes to the kernel driver? Need to investigate.

zxystd, 2021-08-19:
@JJTech0130 No, `owl` projects use active monitor mode to listen and inject frames to communicate with other Apple devices, most of the wireless cards don't support active monitor mode(be careful to distinguish with monitor mode), so does Intel card.
Too much works(P2P client, P2P GO, AWDL, NAN etc.) should be done in driver side and will affect the station mode(current itlwm framework is not satisfied with the works). I don't have too much effort and time to focus on doing these works.

closed 2023-06-11 by zxystd