OpenIntelWireless/itlwm #666: AX200. Airportitlwm stops working after sleep.

InsOnYou, 2021-08-19:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**
- Kext Version: v2.0.0 stable
- WiFi Card Model: AX200
- PCI Product ID: 0x0723    PciRoot(0x0)/Pci(0x1C,0x5)/Pci(0x0,0x0)
- macOS Version: 12 Monterey

**Description**
Airportitlwm works fine, but stops working after sleep. 
It's OK to turn on/off the wifi in settings but no internet can be found.
I've also tried Itlwm and got the same error after sleep.

**Bug Report Archive**
`[  121.494251]: Sandbox: mediaremoted(103) allow iokit-open-user-client IOAudioControlUserClientitlwm: setSCANCACHE_CLEAR
[  123.504539]: PMRD: kIOMessageSystemCapabilityChange[9] to com.apple.iokit.IO80211FamilyLegacy
[  123.504953]: PMRD: kIOMessageSystemCapabilityChange[12] to com.apple.iokit.IO80211FamilyLegacy
[  123.518913]: PMRD: kIOMessageSystemHasPoweredOn to com.apple.iokit.IO80211FamilyLegacy
[  123.519314]: PMRD: kIOMessageSystemHasPoweredOn to com.apple.iokit.IO80211FamilyLegacy
[  123.646262]: itlwm: enable
[  123.646266]: itlwm: iwx_set_hw_ready
[  123.646274]: itlwm: taskq task_add iwx_init_task add to queue
[  123.646292]: itlwm: iwx_init_task
[  123.659074]: itlwm: setIE en1 frame_type_flags 5 add 1 signature_len 12 ie_len 12
[  123.663630]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
[  123.663704]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
[  126.660337]: System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/8243B865FA8E34A32B4E9F35206042BE3BF5C2F4System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/8243B865FA8E34A32B4E9F35206042BE3BF5C2F4itlwm: setSCANCACHE_CLEAR
[  128.056238]: System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.SharedWebCredentials/swc.dbSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.SharedWebCredentials/swc.dbitlwm: setSCANCACHE_CLEAR
[  128.825064]: Sandbox: nsurlsessiond(529) deny(1) mach-lookup com.apple.cfnetwork.AuthBrokerAgentSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E2CEA19AAF3A93C581CC95E0FF39ACA266DC301D/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/F4CA19C5E092DDCE50F2A76C282A3F0974A578BC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E0C1642924E7F6DFC8B9721CA575EDD0CC380294/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/976DF38D04BB325CBC9C10F54546B99B54F9C163/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/976DF38D04BB325CBC9C10F54546B99B54F9C163/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E2CEA19AAF3A93C581CC95E0FF39ACA266DC301D/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E2CEA19AAF3A93C581CC95E0FF39ACA266DC301D/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/CD95FAD77817AF0E294DEAB51B4FAA045DD37EB0System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/CD95FAD77817AF0E294DEAB51B4FAA045DD37EB0/UploadsSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/CD95FAD77817AF0E294DEAB51B4FAA045DD37EB0/configuration.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/CD95FAD77817AF0E294DEAB51B4FAA045DD37EB0/options.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/FEA8602A9C8737E7B92C0E8A3BAC5C4D5CED73EC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/FEA8602A9C8737E7B92C0E8A3BAC5C4D5CED73EC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/FEA8602A9C8737E7B92C0E8A3BAC5C4D5CED73EC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/23799EE6E95D9A3CC914037DDE5DD08A61B5FA40/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/23799EE6E95D9A3CC914037DDE5DD08A61B5FA40/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/sessions.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/sessions.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/CD95FAD77817AF0E294DEAB51B4FAA045DD37EB0/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/F4CA19C5E092DDCE50F2A76C282A3F0974A578BC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/F4CA19C5E092DDCE50F2A76C282A3F0974A578BC/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E0C1642924E7F6DFC8B9721CA575EDD0CC380294/tasks.plistSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E0C1642924E7F6DFC8B9721CA575EDD0CC380294/tasks.plistSandbox: donotdisturbd(428) deny(1) file-write-create /Volumes/Data/.TemporaryItems/folders.501/TemporaryItems/NSIRD_donotdisturbd_SAuivZitlwm: setSCANCACHE_CLEAR
[  129.162321]: System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E2CEA19AAF3A93C581CC95E0FF39ACA266DC301DSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377/E2CEA19AAF3A93C581CC95E0FF39ACA266DC301DSystem Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884E881706B377System Policy: fseventsd(102) deny(1) file-read-metadata /private/var/folders/5t/q3rxwhfj5vl4vyrgmc3ksnhw0000gn/0/com.apple.nsurlsessiond/E8C9768780A371F36651AB699D884 [... cut ...]
[  140.643683]: Sandbox: mobileassetd(251) deny(1) file-read-metadata /Libraryitlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
[  140.769403]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
[  145.769053]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
[  145.769120]: itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
[  147.533492]: itlwm: setSCANCACHE_CLEAR

Time ran: 2021-08-19 23:58:19.1340
Kext not loaded

macOS Version 12.0 (Build 21A5304g)
`

**Kext Download Source**
Downloaded from official release on Github.

williambj1, 2021-08-20:
I'm guessing that you have injected invalid device properties (likely related to ASPM). Closed since this is more like a user mistake. Feel free to comment if that is not a case and please refer to the docs and recollect the logs properly (the one provided is incomplete).

closed 2021-08-20 by williambj1

zxystd, 2021-08-20:
@InsOnYou 
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/7019815/Monterey.zip)
Some platform/ACPI switch wifi off after sleep, so it will not work again. So I manually switch on the wifi again to avoid this kind off ACPI issue, Please try this one to confirm if it is fixed.

InsOnYou, 2021-08-20:
> @InsOnYou
> [Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/7019815/Monterey.zip)
> Some platform switch wifi off after sleep, so it will not work again. So I manually switch on the wifi again to avoid this kind of ACPI issue, Please try this one to confirm if it is fixed.

This version worked. Thank you for your wonderful fix. Maybe I used the wrong SSDT before.

zxystd, 2021-08-20:
@InsOnYou Ok, thank you.

SesshomaruVN, 2021-11-29:
> @InsOnYou
> [Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/7019815/Monterey.zip)
> Some platform switch wifi off after sleep, so it will not work again. So I manually switch on the wifi again to avoid this kind of ACPI issue, Please try this one to confirm if it is fixed.

Hi, I am facing the problem like @InsOnYou, but I am using bigsur instead. Could you tell me what is different between your kext file and 2.1.0 release? Or how to making the fix like your kext file above? Sorry for my bad, because I am not a developer. Look forward to your replying. Thanks.