OpenIntelWireless/itlwm #669: Kernel Panic on 7265 AC

SkyrilHD, 2021-08-22:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: 2.0.0_22a83ab
- WiFi Card Model: Intel Dual Band Wireless-AC 7265
- PCI Product ID: 0x8086 0x095A
- macOS Version: macOS Big Sur 11.5

**Description**

I don't know if it got fixed with newer versions of itlwm since I am running a lower version of itlwm. I woke the system up and saw that I had one bar with no connection. After disconnecting and trying reconnecting, it gave me the spinning thing, indicating it is trying to connect to my wifi router. Since it took a long time to connect, I pressed Fn + WiFi to disable my WiFi on hardware-level (to "reset" my card) and then it panicked. 

It happened very randomly, so I might not be able to reproduce it.

**Bug Report Archive**

```
panic(cpu 3 caller 0xffffff80235b3562): itlwm: nic locks counter 0
Backtrace (CPU 3), Frame : Return Address
0xffffffa0b67938a0 : 0xffffff801f48e04d mach_kernel : _handle_debugger_trap + 0x3fd
0xffffffa0b67938f0 : 0xffffff801f5d4e13 mach_kernel : _kdp_i386_trap + 0x143
0xffffffa0b6793930 : 0xffffff801f5c540a mach_kernel : _kernel_trap + 0x55a
0xffffffa0b6793980 : 0xffffff801f432a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0b67939a0 : 0xffffff801f48d86d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffa0b6793ac0 : 0xffffff801f48db63 mach_kernel : _panic_trap_to_debugger + 0x273
0xffffffa0b6793b30 : 0xffffff801fc9dc0a mach_kernel : _panic + 0x54
0xffffffa0b6793ba0 : 0xffffff80235b3562 com.zxystd.AirportItlwm : __ZN6ItlIwm21iwm_nic_assert_lockedEP9iwm_softc + 0x42
0xffffffa0b6793bc0 : 0xffffff80235b7bce com.zxystd.AirportItlwm : __ZN6ItlIwm14iwm_enable_txqEP9iwm_softciiiiii + 0x8e
0xffffffa0b6793c70 : 0xffffff80235e2f52 com.zxystd.AirportItlwm : __ZN6ItlIwm15iwm_add_aux_staEP9iwm_softc + 0x82
0xffffffa0b6793d00 : 0xffffff80235dc96f com.zxystd.AirportItlwm : __ZN6ItlIwm11iwm_init_hwEP9iwm_softc + 0x2ff
0xffffffa0b6793d70 : 0xffffff80235dbdd6 com.zxystd.AirportItlwm : __ZN6ItlIwm8iwm_initEP6_ifnet + 0xd6
0xffffffa0b6793df0 : 0xffffff80235e0700 com.zxystd.AirportItlwm : __ZN6ItlIwm13iwm_init_taskEPv + 0x110
0xffffffa0b6793e40 : 0xffffff802353a0bf com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x6f
0xffffffa0b6793fa0 : 0xffffff801f43213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.0)[7B21165C-1B86-3896-9EA6-E76549F8A608]@0xffffff8023537000->0xffffff8024292fff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[D9018D72-BFC9-3615-8AA1-30D6A4A80BDF]@0xffffff8020f13000->0xffffff8021092fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[81C90C05-9E8E-30E7-AB18-C559422910E4]@0xffffff8021ce0000->0xffffff8021cf6fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[F7AA4BAF-35CF-3784-A50C-5FD9CDCBE488]@0xffffff8021f6c000->0xffffff8021f94fff

Process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20G71

Kernel version:
Darwin Kernel Version 20.6.0: Wed Jun 23 00:26:31 PDT 2021; root:xnu-7195.141.2~5/RELEASE_X86_64
Kernel UUID: FECBF22B-FBBE-36DE-9664-F12A7DD41D3D
KernelCache slide: 0x000000001f200000
KernelCache base:  0xffffff801f400000
Kernel slide:      0x000000001f210000
Kernel text base:  0xffffff801f410000
__HIB  text base: 0xffffff801f300000
System model name: MacBookPro12,1 (Mac-E43C1C25D4880AD6)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 12245087356234
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000b2308150a2f
  Sleep   : 0x00000a8f8e47ed38 0x0000000083ec918d 0x00000358c5064880
  Wake    : 0x00000a8f938a45ea 0x000000008474e638 0x00000a8f8f1a9059
```

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/commit/22a83ab5e319d8e5a834697accf5069b8981bec7

williambj1, 2021-08-22:
@zxystd

williambj1, 2021-08-22:
Please try the latest version.

SkyrilHD, 2021-09-11:
Closing this issue, as it seems to be fixed in latest versions.

closed 2021-09-11 by SkyrilHD