OpenIntelWireless/itlwm #671: Wi-fi turns on but doesn't show any networks

hanlywijaya, 2021-08-24:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0-stable
- WiFi Card Model: Killer(R) Wi-Fi 6 AX1650i 160MHz Wireless Network Adapter (201NGW)
- PCI Product ID: 0x8086 (vendor)/ 0x34F0 (device)
- macOS Version: 11.5.2-stable

**Description**
I used the same version of AirportItlwm.kext when I installed Big Sur, worked just fine. When I restarted, it didn't work. I've tried everything (all your troubleshooting steps in the manual for troubleshooting) & it doesn't help. I've also tried the version that uses Heliport but no difference unfortunately.

**Bug Report Archive**
Attached. [bugreport_20421.zip](https://github.com/OpenIntelWireless/itlwm/files/7038076/bugreport_20421.zip)
**Kext Download Source**
I downloaded the kext from the Github site (OpenIntelWireless / itlwm) & its a already compiled kext.

zxystd, 2021-08-24:
ACPI related issue, I have do a workaround here e98b1b3cc2b1451594aa4ab6c7fc57af0523427c, you can try 2.1.0-alpha version.

hanlywijaya, 2021-08-24:
> ACPI related issue, I have do a workaround here [e98b1b3](https://github.com/OpenIntelWireless/itlwm/commit/e98b1b3cc2b1451594aa4ab6c7fc57af0523427c), you can try 2.1.0-alpha version.

Sorry if I sound a bit dumb but I'm a bit new to this type of stuff and I don't understand how to do the workaround. Also, I tried the 2.1.0-alpha and same thing happened.

zxystd, 2021-08-24:
If you are confirmed that the newest alpha version can not work with your hack, that you should fix the ACPI yourself, the driver can not wake up the network device so it can not do anything.

hanlywijaya, 2021-08-24:
Alright, will try my best in fixing my ACPI first and I will update when I finish.

hanlywijaya, 2021-08-24:
Fixed it and worked. Sorry for wasting your time and I will close this.

closed 2021-08-24 by hanlywijaya

zxystd, 2021-09-27:
@hanlywijaya Hi, can you help me to check if the latest alpha version also fix the issue?

hanlywijaya, 2021-10-03:
@zxystd Hi, sorry for the late reply. Sure, I can try to fix it. By the way, my Wi-Fi and Bluetooth went out again and after fixing up my ACPI again properly, no change. I'll test the alpha.

hanlywijaya, 2021-10-03:
Just tried your alpha version of AirportItlwm kext and no change. I'm pretty sure its not my ACPI at this point. I'll try to fix it first before continuing.

hanlywijaya, 2021-10-04:
Hi, just tried the new debug of alpha of both ITLWM for Heliport & the Airport version with them both not detecting networks. I've tried fixing my ACPI and remapping my USB with no difference. The kext is running but can't detect networks. I've tested both Monterey & Big Sur with same result unfortunately.

hanlywijaya, 2021-10-04:
Fixed up my wifi and now its working no problem. The debug version of AirportItlwm is working no problem and Wi-Fi is faster than it used to be. @zxystd