OpenIntelWireless/itlwm #674: No connection with AC networks (Failed DHCP)

jcbente, 2021-09-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0 (Release) / 2.1.0-alpha (1e857b9) - Both itlwm+Heliport and Airportitlwm
- WiFi Card Model:  Intel Dual Band Wireless-N 7260 (7260HWM BN)
- PCI Product ID: 08B1
- macOS Version: 11.5.2

**Description**
Since version 2.0.0, connecting to AC Networks fails in properly assigning IP vía DHCP. The system gets an auto-assigned IP, and renewal of DHCP assignment keeps generating a new auto-assigned IP. Assigning an IP manually resolves the assignment alert from macOS, but cannot reach any other ip of the network.  

This behavior only happens when connecting to AC enabled networks (even on N bands) and with this version. Booting off windows with the same card, connects to the full 867Mbps without any issues, discarding any hardware fault. 

Currently running on version 1.3.0 without problems (probably due to the missing AC implementation).

**Bug Report Archive**
[bugreport_50185.zip](https://github.com/OpenIntelWireless/itlwm/files/7130407/bugreport_50185.zip)

**Kext Download Source**
Official Release

elsacorv, 2021-09-22:
Similar here but only for openwrt flashed routers.
With v1.2 it get dhcp ip correctly but wifi keep droping connecting every some seconds.
With v1.3 or 2.0 have same problem but work very well with others routers or android hotspots

Sometimes it gets correctly dhcp ip with v1.3 and wifi work 100% ( in some situations where i connected before to android hotspot and got ip like 192.168.43.xxx )

jcbente, 2021-09-30:
Just tested with the new 2.1.0-alpha (6d52a7c) with both `-novht` and `-noht40`, and i'm able to get IP on 5GHz band with a huge improvement in connection speed vs 1.3.0. 

Hopefully this helps others and giving some idea of what is going on.

MHL3060, 2021-10-11:
where to set the `-novht`?