OpenIntelWireless/itlwm #679: Intel Wireless-AC 9260 not loaded during Recovery Install

Sophie-bear, 2021-09-16:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: v2.0.0 stable
- WiFi Card Model: Intel® Wireless-AC 9260
- PCI Product ID: 0x25268086
- macOS Version: 11.6 (Recovery installer)

**Description**

When I load into the macOS Recovery installer for 11.6, the kext is loaded fine but no wireless adapter is detected. This happens on both AirportItlwm.kext and itlwm.kext
I double checked the AirportItlwm's info.plist for my PCI Device ID and confirmed it is indeed present.
Because of this issue, I am unable to complete the recovery install process over WiFi, which would otherwise be my preferred method of installation. This also prevents me from doing most of the troubleshooting process and providing logs since I'm unable to load into the OS beyond the installer.

The card is adapted to PCIe using a Gigabyte product (https://www.gigabyte.com/Motherboard/GC-WB1733D-I-rev-10#ov)

I have verified through terminal that the PCIe card is detected, and confirmed the Device ID is correct this way.

I do not seem to have this issue on my other PC, which is using an Intel Wi-Fi 6 AX200 160MHz adapter (onboard, MSI MAG X570 Tomahawk WIFI) with AirportItlwm.kext

**Bug Report Archive**

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases
Not custom built.

williambj1, 2021-09-16:
dmesg should work in recovery.

Did it work in non recovery environments before?

Sophie-bear, 2021-09-16:
> 
> 
> dmesg should work in recovery.
> 
> Did it work in non recovery environments before?

I used this same NIC on other systems in the past, albeit with much older versions of itlwm
Here's the log.

[Log_2021-09-16_11-23-25.log](https://github.com/OpenIntelWireless/itlwm/files/7179091/Log_2021-09-16_11-23-25.log)

Sophie-bear, 2021-09-17:
I'm not sure what I did to fix it, but it's working now. A couple things I did all at once and something must have fixed it. 

- Enabled Network Stack in Gigabyte UEFI under the Other OS settings. I figured this was just for booting from LAN but it might do more than that.
- Removed both Intel Bluetooth kexts from OpenCore (for now)

Not sure which did the trick, but it's working now.

closed 2021-09-17 by Sophie-bear