OpenIntelWireless/itlwm #680: AirportItlwm disconnection every 10 minutes with EAPOL messages

richibrics, 2021-09-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.1.0-alpha (happens also on lower versions)
- WiFi Card Model: AC 9560
- PCI Product ID: 0x8086 0x02F0
- macOS Version: 11.5.2 (20G95)

**Description**
Using AiportItlwm every 10 minutes (accurately) I get disconnected from my wireless connection. From macOS console I see EAPOL messages arriving at that exactly moment. 
This doesn't happen with itlwm and HeliPort.

**Bug Report Archive**
[bugreport_25634.zip](https://github.com/OpenIntelWireless/itlwm/files/7209770/bugreport_25634.zip)

**Kext Download Source**
Kext downloaded from release page on this repo, version 2.1.0 (alpha) for macOS Big Sur

richibrics, 2021-09-22:
Additional log of the following day: EAPOL filter in Console at the moment wireless network disconnected 

```
default	12:43:12.396630+0200	kernel	en0: Received EAPOL packet (length = 145)
default	12:43:12.396639+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:12.396653+0200	kernel	inputEAPOLFrame: decrypting key data
default	12:43:12.396661+0200	kernel	inputEAPOLFrame: Received message 1 of 2
default	12:43:13.405646+0200	kernel	en0: Received EAPOL packet (length = 145)
default	12:43:13.405656+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:13.405667+0200	kernel	inputEAPOLFrame: decrypting key data
default	12:43:13.405673+0200	kernel	inputEAPOLFrame: Received message 1 of 2
default	12:43:14.414629+0200	kernel	en0: Received EAPOL packet (length = 145)
default	12:43:14.414638+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:14.414649+0200	kernel	inputEAPOLFrame: decrypting key data
default	12:43:14.414654+0200	kernel	inputEAPOLFrame: Received message 1 of 2
default	12:43:15.424020+0200	kernel	en0: Received EAPOL packet (length = 145)
default	12:43:15.424036+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:15.424060+0200	kernel	inputEAPOLFrame: decrypting key data
default	12:43:15.424071+0200	kernel	inputEAPOLFrame: Received message 1 of 2
default	12:43:39.342941+0200	kernel	en0: Received EAPOL packet (length = 113)
default	12:43:39.342948+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:39.342952+0200	kernel	inputEAPOLFrame: Received message 1 of 4
errore	12:43:39.366072+0200	kernel	Sandbox: rapportd(382) deny(1) mach-lookup com.apple.network.EAPOLController
default	12:43:39.366213+0200	rapportd	eapolcontroller_server_port failed, unknown error code
default	12:43:39.378144+0200	rapportd	eapolcontroller_server_port failed, unknown error code
default	12:43:40.354124+0200	kernel	en0: Received EAPOL packet (length = 113)
default	12:43:40.354140+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:40.354146+0200	kernel	inputEAPOLFrame: Received message 1 of 4
default	12:43:43.275763+0200	kernel	Sandbox: 1 duplicate report for rapportd deny(1) mach-lookup com.apple.network.EAPOLController
default	12:43:45.947854+0200	kernel	en0: Received EAPOL packet (length = 113)
default	12:43:45.947865+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:45.947871+0200	kernel	inputEAPOLFrame: Received message 1 of 4
default	12:43:45.961452+0200	kernel	en0: Received EAPOL packet (length = 209)
default	12:43:45.961460+0200	kernel	inputEAPOLFrame: 0 extra bytes present in EAPOL frame.
default	12:43:45.961472+0200	kernel	inputEAPOLFrame: decrypting key data
default	12:43:45.961483+0200	kernel	inputEAPOLFrame: Received message 3 of 4
default	12:43:45.976425+0200	rapportd	eapolcontroller_server_port failed, unknown error code
errore	12:43:45.976346+0200	kernel	Sandbox: rapportd(382) deny(1) mach-lookup com.apple.network.EAPOLController
default	12:43:45.986525+0200	rapportd	eapolcontroller_server_port failed, unknown error code
default	12:43:45.994282+0200	rapportd	eapolcontroller_server_port failed, unknown error code
default	12:43:46.253803+0200	kernel	Sandbox: 2 duplicate reports for rapportd deny(1) mach-lookup com.apple.network.EAPOLController
```

zxystd, 2021-09-27:
Your router is set to exchange keys every 10 minutes, I don't know why Airport reject to handle this, but I suggest to change the router setting since the keys exchange too frequently will reduce the experience of WiFi network.

usr-sse2, 2022-01-21:
Strange, does it happen on **1.2.0** version? I tested key renew in those times and it worked.

richibrics, 2022-01-21:
Thanks for the tip, I'm testing it now: received correctly 2 messages in 20 minutes. 
I'll update you later.

richibrics, 2022-01-22:
@usr-sse2 I don't have any problem since yesterday when I downgraded AirportItlwm to 1.2.0; hope that it continues like that.
Now the question is: what has changed to key renew in the latest versions ?

zxystd, 2022-03-31:
@riccardo-briccola Hi, I am going to resolve the problem, but I can't reproduce that, can you help me to test?

riccardo-briccola, 2022-03-31:
Sure I am available to test it

zxystd, 2022-03-31:
[Monterey.zip](https://github.com/OpenIntelWireless/itlwm/files/8390886/Monterey.zip)
[Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/8390894/Big.Sur.zip)
I upgrade firmware version to -68, can you see that if it would help?

zxystd, 2022-04-01:
@riccardo-briccola Hello?

riccardo-briccola, 2022-04-01:
Sorry problems with repo watch. I'll test it this afternoon thank you !

riccardo-briccola, 2022-04-01:
Using it for an hour, seems okay; I'll continue testing it and keep writing you. Thank you

riccardo-briccola, 2022-04-01:
Yes same router, this is the report. Sorry for the delay but I had problems with DebugEnhancer that now I've solved
[bugreport_6599.zip](https://github.com/OpenIntelWireless/itlwm/files/8400252/bugreport_6599.zip)
.

zxystd, 2022-04-02:
@riccardo-briccola Thank you for the testing, so now I have push the changes to the repo, you can now use the latest alpha version to test, any issue please contact me.

riccardo-briccola, 2022-04-02:
Fantastic, thank you for the help !

riccardo-briccola, 2022-04-29:
After 27 days of daily use, it has always worked perfectly. Thanks again!

zxystd, 2022-04-29:
Got it! Thank you for the help and confirmation.

richibrics, 2022-10-03:
Works seamlessly since that update. Thank you !!!

closed 2022-10-03 by richibrics