OpenIntelWireless/itlwm #681: DHCP connecting problem OpenWRT

elsacorv, 2021-09-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0 Stable
- WiFi Card Model: Wifi 6 AX201
- PCI Product ID: 
- macOS Version: 11.6 Bigsur

**Description**
Can't get the rigth DHCP IP only with OpenWRT routers (Tested 3 diferent openwrt flashed routers)
Same problem as #https://github.com/OpenIntelWireless/itlwm/issues/584

With itlwm v1.2 it get the rigth dhcp ip and wifi work but keeps connecting/disconnecting in some seconds.
With v1.3 or 2.0 randomly connect and all work fine until restart.

Tested #fc497c1 alpha and latest release alpha but no success.

**Bug Report Archive**
[Log_2021-09-22_22-44-28.log](https://github.com/OpenIntelWireless/itlwm/files/7214697/Log_2021-09-22_22-44-28.log)

**Kext Download Source**
Releases

MaxSYC, 2021-09-23:
@elsacorv Which `OpenWRT` version are you using now? And can you share you wireless security configuration screenshot. I'm curious if this bug is triggered by how we set that. Currently I still can reproduce this problem if I don't use my setting below
[image]
By the way, I'm using `OpenWRT v21.02.0` and this is found when testing #676.

williambj1, 2021-09-23:
@MaxSYC Does WiFi work as expected for you when 11w is disabled? Would you mind sharing the testing results from #676 as well? Thanks.

SnowWarri0r, 2021-09-23:
I have this problem too.But when I use WPA-PSK,it will be OK for me.But it will disconnect and connect over and over again until it gets IP.

SnowWarri0r, 2021-09-23:
> @elsacorv Which `OpenWRT` version are you using now? And can you share you wireless security configuration screenshot. I'm curious if this bug is triggered by how we set that. Currently I still can [...]

I tried this configuration but it didn't work on my laptop.

MaxSYC, 2021-09-23:
@williambj1 Here is my [Logs](https://github.com/OpenIntelWireless/itlwm/files/7218120/Log_2021-09-23.log) (if you still need it). I just following all setting mentioned in 676# and the result is as the same as this DHCP issue, get a self-assigned ip addr. 
With latest alpha release and 11w disable, connection stability and speed is the same under `Windows`.

elsacorv, 2021-09-23:
I tested 3 routers versions are:

- 19.07.7
- 19.07.3
- Lede 17.x

Tried WPA2-PSK and WPA-PSK.
Managment Frame protection is off.

Sometimes with v1.3 when i connect to android hotspot before where mac get ip 192.168.43.17x and after connect to openwrt router worked 2 or 3 times and wifi was 100%

[image]

elsacorv, 2021-09-24:
> @elsacorv Which `OpenWRT` version are you using now? And can you share you wireless security configuration screenshot. I'm curious if this bug is triggered by how we set that. Currently I still can [...]

Upgraded Openwrt to v21..02.0 but no success. ( Whats you router model? )
Can post your router /etc/config/dhcp?

Thanks

MaxSYC, 2021-09-25:
```config dnsmasq
	option domainneeded '1'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option expandhosts '1'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option localservice '1'
	option ednspacket_max '1232'
	option local '/yourname/'
	option domain 'yourname'
	option sequential_ip '1'
	option confdir '/tmp/dnsmasq.d'
	option noresolv '1'
	option doh_backup_noresolv '-1'
	list doh_backup_server ''
	option filterwin2k '1'
	list server '0::1#5054'
	list server '127.0.0.1#5054'

config dhcp 'lan' //*Maybe this part*
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'relay'
	option ndp 'relay'
	option ra 'server'
	option ndproxy_slave '1'
	list ra_flags 'none'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'
	option start '100'
	option limit '150'
	option leasetime '12h'
	list ra_flags 'none'

config odhcpd 'odhcpd'
	option maindhcp '0'
	option leasefile '/tmp/hosts/odhcpd'
	option leasetrigger '/usr/sbin/odhcpd-update'
	option loglevel '4'

config dhcp 'wan6'
	option interface 'wan6'
	option ignore '1'
	list ra_flags 'none'
```
I'm using Mi R3G model(`mt7621` SoC), but I don't think I have made any specific modification to my dhcp config(?) Hoping this can help you anyway.

zxystd, 2021-09-26:
> By the way, I'm using `OpenWRT v21.02.0` and this is found when testing #676.
@MaxSYC Hi, you mean you have seen problems when testing #676 with `11w` enabled? CC @pigworlds

zxystd, 2021-09-26:
@elsacorv with the log, I can see the WiFi signal is weak and trigger the roam scan, can you move closely to the router and test again?

elsacorv, 2021-09-26:
> @elsacorv with the log, I can see the WiFi signal is weak and trigger the roam scan, can you move closely to the router and test again?

There is 0% chance of be for signal strenght ( is very good signal ) but i tested anyway at 3 meters and same result. 
With v1.2 it gets correct dhcp ip in 3 or 4 seconds ( and internet work until 10 or 15 seconds when i do adsl speed test and it start disconect and connect and work for some seconds and repeat).

So i think the problem is some commit v1.2 -> v1.3 related to auth & dhcp, maybe some of these:

- Only retry when current state is AUTH. ASSOCIATION state should be an…
- reject invalid status.
- Give more times to try associate to a BSS.

Maybe can you compile v1.3 without these for test? ( i dunno really just giving out ideas)

@MaxSyc
Thanks for your dhcp config. 
I tested some params in my 19.07.7 because i think maybe is the " default config " of v21.02.0 who is doing the magic but no success. 
Tomorrow i will upgrade again to v21.02.0 but without saving my current config so i can test with default config.

pigworlds, 2021-09-26:
> > By the way, I'm using `OpenWRT v21.02.0` and this is found when testing #676.
> 
> @MaxSYC Hi, you mean you have seen problems when testing #676 with `11w` enabled? CC @pigworlds

@MaxSYC I saw your log indicate your card is ax200 using iwx. Currently the 11w support in #676 doesn't cover any gen2/gen3 card yet. I just updated the description to call it out. What the behavior your see is expected when enable 802.11w in OpenWrt.

@zxystd I don't have any gen2/gen3 card in my hand to test. Do you mind if I enable 11w support for iwx using mixed hw/sw crypto, similar to iwm in #676?

zxystd, 2021-09-27:
@pigworlds 
> @zxystd I don't have any gen2/gen3 card in my hand to test.

Sorry I don't notice it is a iwx device.

>  Do you mind if I enable 11w support for iwx using mixed hw/sw crypto, similar to iwm in #676?

Of course you can do that, and post it here or #676 then somebody who can download to test.

williambj1, 2021-10-02:
@elsacorv Hi! Would you mind sharing your hostapd config file? Thanks!

elsacorv, 2021-10-04:
> @elsacorv Hi! Would you mind sharing your hostapd config file? Thanks!

Yeah for sure

 /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B /var/run/hostapd-phy0.conf

Here is the config:
`
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
beacon_int=100
channel=11

noscan=1

ieee80211n=1
ht_coex=0
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=net24pass
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=Net24
bridge=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=ac:84:c6:57:66:b3

bss=wlan0-1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=vafpass
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=vaf
bridge=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=ae:84:c6:57:66:b3
`

Here is /etc/config/dhcp
`
config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option ra 'server'
        option dhcpv6 'server'
        option ra_management '1'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'
`

elsacorv, 2021-10-05:
@zxystd @williambj1 @pigworlds 

Ok i figured where the problem start
( I already tested beforce with v1.3.0 and 2.0 settings the router at 20mhz and no force 40mhz but dhcp problem is present )

- Compiled v1.3.0 at Merge branch 'ampdu_impl' into abc - f7fd518

Result: is fully working wifi at 20mhz? ( no connect/disconnect like in v1.2.0 ). Max download speed is 55 Mbps ( In windows i get about 70mbps)

- Compiled v1.3.0 at support 40mhz - e4051fb
Result: cant connect and keep asking for wifi password

- Compiled v1.3.0 at enable Tx agg for iwi again. - 7a0e773
Result: cant connect and keep asking for wifi password

-  Compiled v1.3.0 at use correct rate for non-data frames - a6af0af
Result: it connects but got the same problem of dhcp auto assigned ip

-  Compiled v1.3.0 at set iwm rate table 40mhz flag, now the tx speed is normal. - 9d7ba10
Result: it connects but got the same problem of dhcp auto assigned ip

williambj1, 2021-10-05:
@elsacorv When the advanced menu from airportd (hold <kbd>ALT</kbd> while clicking the menu icon) is opened, does it say it's connected with channel 11, 40MHz?

I'm not an expert but I noticed that `ht_capab` has `[HT40-]`. Is it possible to enable the AP manually by passing a config to hostapd from the command line while setting `ht_capab` to `[HT40+]` and removing the hardcoded channel number on your setup? If so, could you try removing `[TX-STBC][RX-STBC1]` as well?

Here is the config from a router running OpenWRT having no issues with itlwm:

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
dtim_period=2
channel=36
chanlist=36

tx_queue_data2_burst=2.0
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[SHORT-GI-80][MU-BEAMFORMEE][MAX-A-MPDU-LEN-EXP0][VHT-LINK-ADAPT-3]

radio_config_id=***
interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=2
wpa_psk_file=/var/run/hostapd-wlan1.psk
eapol_version=1
auth_algs=1
wpa=3
wpa_pairwise=CCMP
ssid=***
bridge=br-lan
snoop_iface=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan1.vlan
config_id=***
bssid=***
```

elsacorv, 2021-11-10:
Tried with Openwrt 21.02.0 default config - itlwm stable 2.0 and no success
Tried with channel 6 which is HT40+ and no success

Last working 100% with compiled v1.3.0 at Merge branch 'ampdu_impl' into abc - f7fd518

zvova7890, 2021-12-17:
Same problem on ax200. I'm installed v1.2.0 seems last working version. OpenWRT 21.02.0 with 40mhz force

elsacorv, 2022-01-23:
> 

Yep, but you can try v1.3.0 compiled at f7fd518.
I had wifi problems and low speed with 1.2.0 and 1.3.0 was not working fine

Check on my user releases
https://github.com/elsacorv/itlwm/releases

zvova7890, 2022-01-25:
I'm checked yours 1.3.0 version and wifi is work! But no speed different between 1.2 and 1.3 in my case. 5Ghz wifi shows around 30mbit with iperf3. I'm using hackintosh in very special cases, so that speed is enought for me

zxystd, 2022-01-27:
@elsacorv @zvova7890 Hi! Have you tried v2.1.0 or v2.2.0-alpha version with boot-args `-noht40`?

williambj1, 2022-03-03:
Closed due to no further response.

closed 2022-03-03 by williambj1

zvova7890, 2022-04-05:
Hi! I have checked v2.1.0 - 2.4 and 5 doesn't work. v2.2.0-alpha - seems work without any tweak, after install both 2.4 and 5 GHz wifi working pretty good, connection speed is much higher than [elsacorv](https://github.com/elsacorv) user build. So I think this is fixed in alpha. Thanks!

zxystd, 2022-04-05:
Yes, there must have something error inside thd -48 firmwares, thank you for the testing.