OpenIntelWireless/itlwm #684: 2.0 和 2.1  局域网上传速度很慢

yinxiulong, 2021-09-24:
我的电脑小米游戏本2019 ，苹果系统为big Sur 11.5   ，用的驱动版2.0 ，wifi 网卡型号为 intel 9560  上传速度很慢，基本在1M--2M 之间最高4M。

inksnw, 2021-09-25:
SMB共享，windows下上传和下载能有70M，mac下大概10M左右，有时会掉到1-2M，也是 intel 9560

williambj1, 2021-09-25:
Invalid issue format. Use iperf for benchmarks and make sure you are sitting within a reasonable distance between the router.

closed 2021-09-25 by williambj1

yinxiulong, 2021-09-25:
3米范围内，速度都慢<html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<br/><br/><br/>
<div style="display: inline-block;"> 
  	<HR width=100% align=left size=0.5 color=#eaebec noshade>
    <font color='#78787A'> 发自 </font>
    <a href='https://mail.163.com/dashi/dlpro.html?from=mail81' target='_blank' style='text-decoration: none; color: #111112'>网易邮箱大师</a>
</div>
<br/><br/><br/><div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 2em; margin-bottom: 1em; font-size: 14px; line-height: 1.25; color: #78787a; " >---- 回复的原邮件 ----</div>
    <div style=" margin-bottom: 1em; font-size: 12px; line-height: 1.25; color: #232324; padding: 0.5em 0.25em; border-radius: 0.67em; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="9" border="0">
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    发件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-from" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    日期
                </td>
                <td class="mail-date" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    2021年09月25日 22:03
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    收件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-to" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    抄送至
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>、<a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    主题
                </td>
                <td class="mail-subject" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    Re: [OpenIntelWireless/itlwm] 2.0 和 2.1  局域网上传速度很慢 (#684)
                </td>
            </tr>
        </table>
    </div>
    <div><p></p>
<p>Invalid issue format. Use iperf for benchmarks and make sure you are sitting within a reasonable distance between the router.</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br />
[image]</p>
<script type="application/ld+json">[
{
"target": "https://github.com/OpenIntelWireless/itlwm/issues/684#issuecomment-927126052",
"url": "https://github.com/OpenIntelWireless/itlwm/issues/684#issuecomment-927126052",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script></div>
</div>
</body>
</html>

zxystd, 2021-09-25:
iperf3
[image]

SMB
[image]

speedtest
[image]

SuperPDX, 2021-10-19:
同样的问题，我的是突然变得巨慢，但是之前是正常的，似乎是升级了opencore以后出现的问题。但是现在也不能确定了