OpenIntelWireless/itlwm #688: Airportitlwm and itwlm lost connection every 5-10 minutes

m3XS, 2021-09-29:
**Have You Read Our Docs**

yes

**Are You Reporting A Bug**

no

**Environment**

- Kext Version: 2.0.0 stable (Big Sur)
- WiFi Card Model: Intel(R) Wireless-AC 9560 160MHz
- PCI Product ID: 
- macOS Version: Big Sur 11.2

**Description**

I try to install Big Sur 11.2 on my Razer Blade 15 Base Model late 2020 with the intel wifi card. Everything works just after 20% Download of InstallAssistant.pkg it aborts the download because i disconnect with my 2.4 GHz connection and reconnect. I cant install Big Sur. I already tried with 5GHz and also 2.4 GHz connection. On my Desktop-PC i could install on the same time.

I tried with Airportitlwm and itlwm with preconfigured wifi configuration.

log file: https://pastebin.com/TbDdZQ7X

**Bug Report Archive**

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/download/v2.0.0/AirportItlwm_v2.0.0_stable_BigSur.kext.zip
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.0.0/itlwm_v2.0.0_stable.kext.zip

QuanTrieuPCYT, 2021-09-30:
Same issue over here. Also after the issue happened, i tried to restart the PC but it hang. Force restarting it caused macOS to show "Your computer restarted because of a problem" on the next boot. Maybe a kernel panic?

williambj1, 2021-10-02:
@m3XS The log file seems to be incomplete, please use the correct keyword for filtering, instructions can be found in the docs.

Have you tried increasing the key expiring interval in the router settings?

@QuanTrieuPCYT Please attach the panic log in new issue since it does not appear to be the same as this one.

QuanTrieuPCYT, 2021-10-02:
okay

closed 2021-10-09 by m3XS