OpenIntelWireless/itlwm #689: Error configuring antenna diversity

Cheesecake8069, 2021-09-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: all
- WiFi Card Model: 7265AC
- PCI Product ID: 0X095A
- macOS Version: CATALINA 10.15.7
- bootloader: opencore

**Description**
So I've been battling this bug for over 2 weeks now, everything works, the kext loads 80211 kexts load and stay loaded, everything surface level works fine, however when you go to look for wifi networks, nothing shows up. After quite a bit of digging through my boot logs, I've found

"Itlwm: tasks task_add init_task add to queue 
en1:setting diversity to: -1
en1: Error configuring antenna diversity (index = -1)
Itlwm: iwm_init_task
en1: setting tx antenna: -1
en1: Error configuring transmit antenna (index = -1)"

**Bug Report Archive**
It literally just says wifi on wifi off create diagnostic report. 
I can't drag and drop it in creating this report from my phone. 

**Kext Download Source**
GitHub official stable.

Cheesecake8069, 2021-09-30:
[image]

zxystd, 2021-09-30:
Every airport interface will print this log, including Apple Broadcom drivers, if you think it is a bug, please fire to Apple.

Cheesecake8069, 2021-09-30:
> Every airport interface will print this log, including Apple Broadcom drivers, if you think it is a bug, please fire to Apple.

So something else I just noticed, when loading the card itlwm states that L1 is disabled and ltr is enabled, 
Buy in hackintool the aspm of the 7265ac is L1. 
I don't think taking this issue to apple is great idea as I'm not sure how in favour they are of using third party kexts through opencource bootloader's...

closed 2021-10-01 by Cheesecake8069