OpenIntelWireless/itlwm #690: no networks

Cheesecake8069, 2021-10-01:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version: All
- WiFi Card Model: 7265AC
- PCI Product ID: 0x095a
- macOS Version: catalina 10.15.7

**Description**
the kext can see what card I'm using, the wifi infinity searches for networks but none ever show up, this was not an issue with clover, but since doing a clean install using open core 6.7.2 i haven't been able to connect to a single wifi network.

**Bug Report Archive**
GitHub won't allow me to upload the wifi diagnostics because its the incorrect file type?
**Kext Download Source**
official GitHub

Cheesecake8069, 2021-10-01:
[image]

Cheesecake8069, 2021-10-01:
new symptom now m wifi won't even friggin turn on....... ergghhhhhhhhhh

williambj1, 2021-10-01:
Sorry but this is not the right place for EFI related issues.

closed 2021-10-01 by williambj1

Cheesecake8069, 2021-10-01:
I would love to know what user mistake I've made, my apple secure boot is enabled(j680).  sanity checker says my config.plist is fine. All my kexts load and report correctly. I just can't see networks.