OpenIntelWireless/itlwm #691: Universal clipboard and Handoff only working one way on Monterey

samwzlim, 2021-10-01:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0
- WiFi Card Model: Intel Dual Band Wireless-AC 8265/8275
- PCI Product ID: 
- macOS Version: 12.0 Beta 6 (21A5506j)

**Description**
Universal clipboard and Handoff only works one way for me (e.g. iPhone to Mac but not Mac to iPhone). I'm sure my hackintosh is the culprit as it works flawlessly between my iPhone, iPad and Apple Watch. Haven't heard of anyone else running Monterey with this kext facing this issue so I'm not sure if it's a Monterey issue or kext issue. Internal wifi and Bluetooth is working as expected on my machine. On Catalina and Big Sur, universal clipboard and handoff were working both ways for me.

Bluetooth kext used: IntelBluetoothFimware.kext coupled with BlueToolFixup.kext

**Kext Download Source**
Official release from this repository

williambj1, 2021-10-01:
This project does not contain any patches that intentionally enables Universal Clipboard or Handoff. Both features work on my laptop running macOS Monterey Developer Beta 8. Have you tried re-logging in iCloud, cleaning up keychains, etc?

Closed since it doesn't appear to be an issue on our side (Userspace issue, unrelated to the kernel).

closed 2021-10-01 by williambj1

samwzlim, 2021-10-01:
> This project does not contain any patches that intentionally enables Universal Clipboard or Handoff. Both features work as expected on my laptop running macOS Monterey Developer Beta 8. Have you tried re-logging in iCloud, cleaning up keychains, etc?
> 
> 
> 
> Closed since it doesn't appear to be an issue on our side (Userspace issue, unrelated to the kernel).

Yes, I have already tried every possible troubleshooting method on different forums but still no luck. Hopefully when the stable release of Monterey comes out, this problem will be resolved (or at least narrow down the cause of issue.