OpenIntelWireless/itlwm #692: Bluetooth low range issues

fredimartins, 2021-10-01:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0 Stable
- WiFi Card Model: Intel Dual Band Wireless-AC 3165 Plus Bluetooth
- PCI Product ID: 0x8087 0x0A2A
- macOS Version: BigSur 11.6

**Description**
When i using only IntelBluetoothFirmware bluetooth works great, 2 meters without problem. When i add itwlm bluetooth problems start, if headphones are more than 20cm i have audio problems.
Whe i using Airportitwlm, same problem, 20cm is the max range of bluetooth but if i turn wifi off, bluetooth works great again. 

**Kext Download Source**
github release

Overc1ocker, 2021-10-05:
Can you potentially update the title of this bug to reference the problematic bluetooth device ids. Right now i'm only seeing cards with the bluetooth device 0x8087, 0x0a2a having issues.

fredimartins, 2021-10-06:
> Can you potentially update the title of this bug to reference the problematic bluetooth device ids. Right now i'm only seeing cards with the bluetooth device 0x8087, 0x0a2a having issues.

done

pigworlds, 2021-10-10:
I finally got a chance access to a NUC with AC7265 card on the board. The wifi pci id is 8086, 095a. the bt usb is 8087, 0a2a.
While AirportItlwm works fine on the box, I do exhibit the same bluetooth issue. I tested with bluetooth headset. I need to hold the headset very close to the box to connect while wifi is on.

I saw @zxystd have posted a test kext [ac7265_enable_coex_flags.zip](https://github.com/OpenIntelWireless/itlwm/files/6867779/ac7265_enable_coex_flags.zip) on https://github.com/OpenIntelWireless/itlwm/issues/482#issuecomment-885513426. It appears the bt coex issue is indeed the cause.

By using the kext, I can use both wifi and bluetooth together. I have tested wifi on both 2.4GHz and 5GHz channels, they're all working fine while playing audio using bluetooth. The 2.4GHz channel speed is slow but at least works. The 5GHz channel works great.

pigworlds, 2021-10-10:
@zxystd I saw the bt coex mode in itlwm and openbsd are both set to 

> `bt_cmd.mode = htole32(IWM_BT_COEX_WIFI);`

In Linux, it is set to

> `bt_cmd.mode = cpu_to_le32(BT_COEX_NW);`

I am guessing `IWM_BT_COEX_WIFI` means prioritize to WIFI, and it causes bluetooth lose opportunity to use antenna.

It appears the setting to `BT_COEX_WIFI` doesn't cause issue to bluetooth on 8xxx and above cards, and the issue only happen on the 7xxx cards. I am not sure why the bt coex value was set to `BT_COEX_WIFI` in the first place. Maybe we should align with Linux, or at least on the 7xxx cards.

I also notice the `iwm_send_bt_init_conf` is called twice. Once inside `iwm_run_init_mvm_ucode`, Another inside `iwm_init_hw` which it calls `iwm_run_init_mvm_ucode`. Not sure if it is redundant or it is a bug.

fredimartins, 2021-10-10:
exacly same behavior here. with that kext i can use bluetooth and wifi at the same time.
With 2.4 GHZ i see random disconnects of BT when i makin speed tests and slow speed 
30Mb in 2.4 and 300Mb in 5ghz.

FammasMaz, 2021-10-10:
Exactly the same issue on 3165ac, same device id. Looks like its finally gonna be solved!

Overc1ocker, 2021-10-11:
Sounds like Arch Wiki suggests a similar fix.
[Bluetooth Coex Intel Arch](https://wiki.archlinux.org/title/Network_configuration/Wireless#Bluetooth_coexistence)

singhalrishi27, 2021-10-14:
yup same issue with 0x8087 0x07DC when bt and wifi together

alikaylan, 2021-10-17:
[ac7265_enable_coex_flags.zip](https://github.com/OpenIntelWireless/itlwm/files/6867779/ac7265_enable_coex_flags.zip) definitely solves the issue regarding using BT while wifi is on. BT Audio works just fine with ample range. 0x8086 wifi. 7265. Would it be possible that this is incorporated into future releases for this setting for 7265?

alikaylan, 2021-10-30:
I have no issues here with the alpha version. WiFi and BT seem to be working together in coexistance just fine on 0x8086 wifi. 7265. BT range is fine. No issues apparent with WiFi. Thank you so much for addressing this issue. Made all the difference in the world. 👍

rboldini, 2021-11-02:
For who has problem with bluetooth audio, some BIOSes provide an option under HD Audio settings  to enable A2DP, this fixed discontinuity during audio play in both windows and macOS.
Device AC7265 BT: 0x0a2a

closed 2022-01-04 by zxystd