OpenIntelWireless/itlwm #695: 7265AC NOT WORKING. TESTED CONFIRMED

Cheesecake8069, 2021-10-04:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: ALL
- WiFi Card Model: 7265AC
- PCI Product ID: 095a
- macOS Version: Catalina 10.15.7

**Description**

The card will not engage. Will not actually turn on although airport itlwm says it's on. Itlwm says it's off when tested with that. Unable to turn it on. 
Also wireless 3160 AC Works perfectly on itlwm despite it not being listed as compatible. Works with airportitlwm too.
Ioname 8b4
Device I'd 0x8270
Aspm Los+L1

I think the current issue with the 7265ac card I have is it gets registered to L1 ONLY and won't engage because of that. 

Personal note: I'm a little mad I was told this issue was user error and a problem with my efi on here a couple times now. 
But it's clearly not an issue with my efi if the 3160AC worked right out of the box. 
I wasted like 2 weeks of my life on this trying to figure out where I screwed up. 
Abit salty. 
But I still appreciate all the work you guys are doing and if you need anyone to write any documentation for the itlwm stuff let me know because at this point I'm not sure there are many other users who have as much troubleshooting experience as I do.....

**Bug Report Archive**
itlwm: iwm_attach
itlwm: , 
itlwm: alloc contig
itlwm: config ieee80211
itlwm: taskq taskq_create_thread lock
itlwm: taskq taskq_create_thread lock
itlwm: taskq taskq_create_thread unlock
itlwm: ieee80211_channel_init
itlwm: ieee80211_set_link_state nstate=2, old_state=-1
itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
itlwm: ieee80211_media_init
itlwm: iwm_preinit
itlwm: iwm_prepare_card_hw
itlwm: iwm_start_hw
itlwm: iwm_prepare_card_hw
itlwm: taskq taskq_create_thread lock
itlwm: taskq taskq_create_thread unlock
itlwm: taskq taskq_create_thread unlock
itlwm: taskq taskq_create_thread lock
itlwm: taskq taskq_thread schedule task
itlwm: taskq taskq_thread schedule task
itlwm: iwm_apm_init
itlwm: itlwm: L1 Disabled - LTR Enabled
itlwm: iwm_init_msix_hw
itlwm: iwm_conf_msix_hw
itlwm: iwm_run_init_mvm_ucode justnvm=1
itlwm: iwm_load_ucode_wait_alive ucode_type=1
itlwm: taskq taskq_create_thread unlock
itlwm: load firmware iwm-7265-17 done
itlwm: iwm_start_fw ucode_type=1
itlwm: iwm_apm_init
itlwm: itlwm: L1 Disabled - LTR Enabled
itlwm: iwm_load_firmware
itlwm: iwm_post_alive
itlwm: iwm_enable_ac_txq
itlwm: iwm_send_bt_init_conf
itlwm: iwm_nvm_init
itlwm: iwm_nvm_read_section
itlwm: iwm_nvm_read_section
itlwm: iwm_nvm_read_section
[... 35 lines cut ...]
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
itlwm: getSYNC_ENABLED
itlwm: setROAM_PROFILE cnt=1 flags=4
itlwm: setROAM_PROFILE cnt=1 flags=2
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
itlwm: setSCANCACHE_CLEAR
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
Sandbox: rapportd(339) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: rapportd(339) deny(1) mach-lookup com.apple.lsd.mapdbSandbox: rapportd(339) System Policy: deny(1) file-read-data /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-1082-v2.csstoreSandbox: rapportd(339) System Policy: deny(1) file-read-metadata /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-1082-v2.csstoreSandbox: 2 duplicate reports for rapportd deny(1) file-read-metadata /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-1082-v2.csstoreSandbox: rapportd(339) deny(1) mach-lookup com.apple.lsd.modifydbSandbox: 70 duplicate reports for rapportd deny(1) mach-lookup com.apple.lsd.modifydbSandbox: rapportd(339) deny(1) mach-lookup com.apple.cdp.daemonSandbox: fontworker(368) deny(1) file-read-xattr /Library/Fonts/Arial Unicode.ttfSandbox: fontd(366) deny(1) file-issue-extension target:/System/Library/AssetsV2/com_apple_MobileAsset_Font6 class:com.apple.app-sandbox.readSandbox: cfprefsd(152) System Policy: deny(1) file-read-data /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.routined/dv/Cache/defaults.plistSandbox: 3 duplicate reports for cfprefsd deny(1) file-read-data /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.routined/dv/Cache/defaults.plistSandbox: mds(114) System Policy: deny(1) file-read-metadata /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/0/com.apple.ScreenTimeAgent/Store/RMAdminStore-Local.sqlitesandboxd rejected approval request from UserEventAgent for kTCCServiceSystemPolicyAllFiles (/Users/grant/Library/Safari): deniedSandbox: UserEventAgent(323) System Policy: deny(1) file-read-data /Users/grant/Library/SafariSandbox: 1 duplicate report for UserEventAgent deny(1) file-read-data /Users/grant/Library/SafariSandbox: UserEventAgent(323) System Policy: deny(1) file-read-data /Library/Application Support/com.apple.TCCSandbox: 1 duplicate report for UserEventAgent deny(1) file-read-data /Library/Application Support/com.apple.TCCSandbox: CommCenter(377) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: CommCenter(377) deny(1) file-read-data /private/var/db/mds/messages/501/se_SecurityMessagesitlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
Sandbox: callservicesd(386) deny(1) file-read-data /Users/grant/Library/Preferences/com.apple.iChat.plistSandbox: callservicesd(386) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: ContextStoreAgen(380) deny(1) ipc-posix-shm-write-create ContextStoreAgentSandbox: ContextStoreAgen(380) deny(1) mach-lookup com.apple.SystemConfiguration.configdSandbox: cloudd(333) deny(1) system-fsctl _IO('h', 47)Sandbox: fmfd(391) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: fmfd(391) deny(1) file-write-data /Users/grant/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: 1 duplicate report for fmfd deny(1) file-write-data /Users/grant/Library/Preferences/com.apple.icloud.fmfd.notbackedup.plistSandbox: networkservicepr(375) deny(1) mach-lookup com.apple.securityd.xpcSandbox: networkservicepr(375) deny(1) file-write-unlink /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/C/mds/mdsObject.dbSandbox: 1 duplicate report for networkservicepr deny(1) file-write-unlink /private/var/folders/6k/bf8cm0gx3qv4n75jsx56qqzr0000gn/C/mds/mdsObject.dbSandbox: apsd(123) deny(1) mach-lookup com.apple.imagent.apsresponsibility for com.apple.dock.e[396] set to that of com.apple.dock.e[396]: /System/Library/CoreServices/Dock.app/Contents/XPCServices/com.apple.dock.extra.xpc/Contents/MacOS/com.apple.dock.extraSandbox: iconservicesd(115) deny(1) process-info-pidinfo selfitlwm: getSYNC_ENABLED
successfully parsed 2/4 entries for file Beautiful mess.logicxsuccessfully parsed 2/4 entries for file Desktopsuccessfully parsed 2/4 entries for file config.plistsuccessfully parsed 1/4 entries for file AirportItlwm.kextsuccessfully parsed 2/4 entries for file Screenshot 2021-10-01 at 14.32.33.pngitlwm: setSCANCACHE_CLEAR
itlwm: setSCANCACHE_CLEAR
itlwm: setSCANCACHE_CLEAR
Waking up reference: 1Thread waiting on reference 1 woke upevaluation result: 1, allowed, cache,  1633197378itlwm: setSCANCACHE_CLEAR
Sandbox: 3 duplicate reports for CalNCService deny(1) file-link /Users/grant/Library/Application Support/AddressBook/.AddressBook-v22_SUPPORT/_EXTERNAL_DATA/C508579A-47EB-4B2F-9C61-73BC2716A37ASandbox: bluetoothd(143) deny(1) mach-lookup com.apple.server.bluetoothitlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) get
itlwm: apple80211Request Unhandled IOCTL BTCOEX_PROFILES (221) set
Sandbox: com.apple.Safari(462) deny(1) file-read-data /Users/grant/Library/SafariSandbox: com.apple.WebKit(465) deny(1) file-write-create /Users/grant/Library/Caches/com.apple.WebKit.NetworkingSandbox: com.apple.WebKit(464) allow iokit-open RootDomainUserClientSandbox: 1 duplicate report for com.apple.WebKit allow iokit-open RootDomainUserClientSandbox: com.apple.WebKit(464) allow mach-lookup com.apple.awddSandbox: com.apple.WebKit(464) deny(1) mach-lookup com [... cut ...] r file Assetssuccessfully parsed 1/4 entries for file FrameworkCachessuccessfully parsed 1/4 entries for file 708e49fabf083cb1251012ad1d8f97303f32a83asuccessfully parsed 1/4 entries for file Recordssuccessfully parsed 1/4 entries for file A795A3E63BA11E4EA2C866DEE63367A7B4522A98-blobsuccessfully parsed 1/4 entries for file F82DA699F69A182E39222816B4F35CDAEA6820C4-blobsuccessfully parsed 1/4 entries for file 8D1162259B27CE60F6B8939D8CF425B

**Kext Download Source**
GitHub official downloads only.

Overc1ocker, 2021-10-05:
I have a 7265AC on MacOS Big Sur and usually get full speeds using the latest AirportITLWM kext. Any reasons for not using big sur or higher?

Cheesecake8069, 2021-10-05:
> I have a 7265AC on MacOS Big Sur and usually get full speeds using the latest AirportITLWM kext. Any reasons for not using big sur or higher?

Yeah I'm not a fan of the user interface. Also compatibility is a mess with some hardware and it still has alot of issues. 
Also not all the 7265AC have the game firmware. There's a freaking ton of different revisions for it. Some of the cards work fine. That's why I included my hardware revision in my log.

Overc1ocker, 2021-10-05:
Yep that's it. I have a "Pci8086,95B" in IOregistryexplorer. I'm assuming that means the card is a rev B

Cheesecake8069, 2021-10-05:
> Yep that's it. I have a "Pci8086,95B" in IOregistryexplorer. I'm assuming that means the card is a rev B

Yeah that's why your card works and mine doesn't haha, mine used to work on one of the really early alpha builds, I don't have a copy of that build anymore and it's not listed in assets. So I couldn't just roll back,

williambj1, 2021-10-06:
@Cheesecake8069 Hi! Looks like the card did not enter the correct state (no errors shown in the logs however) after the firmware was uploaded. No extra hints were provided in the logs unfortunately. Here are some questions that might help finding the cause:

- Does the card work if directly rebooted from Linux?
- Do you have additional PCI properties injected under the pci address in the bootloader's config?
- Does the ASPM info appear the same if the 3160AC is installed instead?

I'm sorry to say that @zxystd is extremely busy these days and he might not be able to investigate this issue soon, we appreciate your patience. Thank you very much.

zfdx123, 2021-10-07:
> > @Cheesecake8069 Hi! Looks like the card did not enter the correct state (no errors shown in the logs however) after the firmware was uploaded. No extra hints were provided in the logs unfortunately. Here are some questions that might help finding the cause:
> > 
> > * Does the card work if directly rebooted from Linux?
> > * Do you have additional PCI properties injected under the pci address in the bootloader's config?
> > * Does the ASPM info appear the same if the 3160AC is installed instead?
> > 
> > I'm sorry to say that @zxystd is extremely busy these days and he might not be able to investigate this issue soon, we appreciate your patience. Thank you very much.
> 
> I'll have to test the Linux thing today as I don't currently have a Linux distro on my laptop. So I'm unsure on that one.
> 
> No there are no other properties being injected.
> 
> The aspm info for the 3160AC appears as L1+LOS, THE 7265AC Only appears as L1 I can upload a log for comparison on the 3160ac if you would like? They are both 7 series cards I believe

Does your BIOS have a whitelist?

Cheesecake8069, 2021-10-07:
> > I'll have to test the Linux thing today as I don't currently have a Linux distro on my laptop. So I'm unsure on that one.
> 
> A live USB would be sufficient.
> 
> > I can upload a log for comparison on the 3160ac if you would like?
> 
> Might not be that useful so maybe not

Okay, so I'm gonna be honest I've never used Linux before so this is going to take me like maybe 2 hours to learn, I'm currently in the process of creating a bootable USB of CentOS, the download from the mirror isn't very fast, so I'm going to try the official torrent. I'll get back to to
You either tonight or tomorrow with the results

pigworlds, 2021-10-12:
> @pigworlds You said you got access to a PCI8086,095A (AC7265) @Cheesecake8069 was actually having problems with their card. Did you happen to notice whether your card is working?

The ac7265 card I have access is in an Intel NUC. It is soldered on the motherboard with 7265D2W on it. The card works well with the latest version of AirportItlwm. I haven't encounter any issue.