OpenIntelWireless/itlwm #705: AC7260 Can't Connect to Network And Can't Refresh Wifi AP List

coolderin, 2021-10-30:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0 and 2.1.0 alpha
- WiFi Card Model: AC7260
- PCI Product ID: (08B1)
- macOS Version: Big Sur 11.6 20G165

**Description**
I can't connect to my 5G and 2.4G wireless access points. It (Wifi) can't refresh network list but if i turn off&on my wifi it can refresh network list once.

**Bug Report Archive**
Wifi Diagnostics: https://drive.google.com/drive/folders/1Czf-VAPb3bVn-Y_YHqKVE95J93PVhiXf?usp=sharing
Log: [Log_2021-10-30_09-56-51.log](https://github.com/OpenIntelWireless/itlwm/files/7446005/Log_2021-10-30_09-56-51.log)

**Kext Download Source**
https://drive.google.com/file/d/1N09OzzkgUaVoAqNKX8TvJ5o8uHxL-33Y/view?usp=sharing(2.0 and 2.1 alpha)

DNCTrung, 2021-11-07:
Same to me with Wireless 3160 after updated. I have to get back 1.3.0. It's working.

zxystd, 2021-11-07:
> Same to me with Wireless 3160 after updated. I have to get back 1.3.0. It's working.

Have you tried v2.1.0-alpha?

zxystd, 2021-11-07:
@coolderin Maybe you can try change to another ssid? Or can your try if it can connect with cellphone hotspot?

coolderin, 2021-11-07:
I tried to connect my cellphone hotspot (2.4G)  and changed my ssid (5G) (eg. "abcd efg" to "abcdefg") it didn't work. I tried on version 2.0.

zxystd, 2021-11-07:
@coolderin what about itlwm.kext? if error still exist, please collect the log again. should be noticed that ac7260 have an connection issue on version 2.0.0, fixed in v2.1.0-alpha.

coolderin, 2021-11-07:
[itlwm=heliport.log](https://github.com/OpenIntelWireless/itlwm/files/7492606/itlwm.heliport.log)
Here is itlwm.kext log ver 2.1.0-alpha. Both of kexts (Airport and Heliport) 2.4Ghz network was connected once.
And System Information>WiFi section my Wifi device ID was 4470 but my device id is 0B81.

zxystd, 2021-11-09:
[image]

why putting itlwm.kext and AirportItlwm.kext together?

> And System Information>WiFi section my Wifi device ID was 4470 but my device id is 0B81.

0x4470 is subdevice id, not device id. I have ac7260 with id 0x0B82 and it works fine.

coolderin, 2021-11-09:
[last-itlwm-notairport.log](https://github.com/OpenIntelWireless/itlwm/files/7506146/last-itlwm-notairport.log)

Here is the "no airport" log. But itlwm disabled the airportitlwm. Is there any problem?

zxystd, 2021-11-09:
@coolderin what's the meaning of 'itlwm disabled the airportitlwm'? I can see the card is init/connect succeed, but sometimes it acquires device fail, seems that something is racing the ownership of the card.

coolderin, 2021-11-09:
I mean when I use airportitlwm and itlwm at same times, itlwm block the airportitlwm. Now I try only itlwm with Heliport, I can't connect to my APs.

williambj1, 2021-11-10:
What makes you think they can coexist with each other? Does your imaginary "blocking" mechanism work in real life? Closed since this is an obvious user mistake, read the docs and use the products as intended.

closed 2021-11-10 by williambj1

coolderin, 2021-11-10:
No sir, I didn't make mistake,
In first log (https://github.com/OpenIntelWireless/itlwm/files/7446005/Log_2021-10-30_09-56-51.log) I didn't use itlwm, I used airportitlwm.
When developer said did you try itlwm with heliport, i tried itlwm with airportitlwm (itlwm already block airportitlwm and itlwm be active), then https://github.com/OpenIntelWireless/itlwm/files/7492606/itlwm.heliport.log i sent log, dev said try with only itlwm then i sent only (https://github.com/OpenIntelWireless/itlwm/files/7506146/last-itlwm-notairport.log) log.
So, I didn't make mistake (maybe i did in 2nd log.)
Please don't close this topic.

jssunhaimin, 2021-11-16:
我的是Intel 7260， Macos 10.15.7的环境， 使用的是2.0驱动 连WIFI 2.4G正常。

连接WIFI 5G没有数据，WIFI图标上有个感叹号！ 目前日常使用的是WIFI2.4G