OpenIntelWireless/itlwm #709: big sur airportitlwm突然没有wifi列表，按照内容重置也没有用

tja414312570, 2021-11-09:
英语不好，我可以用中文吗，就是bigsur  之前还好好的，突然wifi断了，然后wifi列表一直是空的，重置，换itlwm(之前是airportitlwm)+heliport都没用。**下面是系统的wifi日志**
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/7505809/wifi.log)
**以下是wifi信息**
  软件版本：
  CoreWLAN：	16.0 (1657)
  CoreWLANKit：	16.0 (1657)
  附加菜单：	17.0 (1728)
  系统信息：	15.0 (1502)
  IO80211系列：	12.0 (1200.12.2b1)
  诊断：	11.0 (1163)
  AirPort实用工具：	6.3.9 (639.15)
  接口：
en1：
  卡类型：	AirPort Extreme  (0x8086, 0x5010)
  固件版本：	itlwm: 2.0.0 fw: 17.3216344376.0
  MAC地址：	88:78:73:64:64:b4
  位置：	FCC
  国家/地区代码：	ZZ
  支持的PHY模式：	802.11 a/b/g/n/ac
  支持的频道：	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165
  无线网络唤醒：	支持
  隔空投送：	支持
  “隔空投送”频道：	0
  自动解锁：	支持
  状态：	未关联
**以下是desmg的日志**
[Log_2021-11-09_23-13-13.log](https://github.com/OpenIntelWireless/itlwm/files/7505842/Log_2021-11-09_23-13-13.log)

zxystd, 2021-11-09:
Your Wireless card is lock, fix your EFI first. And you can try v2.1.0-alpha.

closed 2021-11-09 by zxystd

tja414312570, 2021-11-13:
我怀疑是wifi模块坏了，但是win10下一切正常，efi和系统都重置过，请问您所说的无线网卡锁是指的什么呢

holoto, 2021-11-13:
wifi国家代码 更改试试
重新打开wifi 开关试试

LeoYule, 2021-12-21:
我也遇到了同样的问题 国家码那里显示的ZZ 和你的一样 请问解决了吗？我也折腾了好久都不行