OpenIntelWireless/itlwm #710: AC7260 Can't Connect to Network And Can't Refresh Wifi AP List (Part 2)

coolderin, 2021-11-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0 and 2.1 (itlwm and airportitlwm)
- WiFi Card Model: AC7260
- PCI Product ID: 0B81
- macOS Version: Big Sur 11.6 20G165

**Description**
First, I didn't make mistake. I sent 3 different logs to you (1-AirportItlwm 2-Itlwm and AirportItlwm 3-Itlwm). Maybe I did mistake in 2nd log. Anyway, I tried all ways. I don't understand. Anyway, I have same issues. It can't connect to Internet, It sometimes can't see 5G 64th channel AP. It can't refresh the AP List. 

**Bug Report Archive**
Only AirportItlwm: https://github.com/OpenIntelWireless/itlwm/files/7446005/Log_2021-10-30_09-56-51.log
AirportItlwm: https://drive.google.com/drive/folders/1Czf-VAPb3bVn-Y_YHqKVE95J93PVhiXf?usp=sharing
Itlwm + Heliport: https://github.com/OpenIntelWireless/itlwm/files/7506146/last-itlwm-notairport.log

**Kext Download Source**
From here.
 

Peace

coolderin, 2021-11-12:
Now I tried again with Heliport It connect 5G netowrk, It had very slow connection and disconnected. 
Log:
[Log_2021-11-12_23-21-37.log](https://github.com/OpenIntelWireless/itlwm/files/7530234/Log_2021-11-12_23-21-37.log)

zxystd, 2021-11-14:
Your card is losing ownership of this driver after while, I don't know what was happened on your machine but it is not our side to fix it, it is out of our control.
[image]

closed 2021-11-14 by zxystd

jssunhaimin, 2021-11-16:
我的是Intel 7260，  Macos 10.15.7的环境，  连WIFI 2.4G正常。

连接WIFI 5G没有数据，WIFI图标上有个感叹号！  目前日常使用的是WIFI2.4G

zxystd, 2021-11-16:
@jssunhaimin use v2.1.0-alpha version instead.

jssunhaimin, 2021-11-16:
> @jssunhaimin use v2.1.0-alpha version instead.

刚刚已经试过，结果相同。
不过刚发现另一个情况，我家里用了两种 TPlink的86型无线ap面板 ：一种是TL-AP1202GI-PoE，另一种是TL-AP1300GI-POE 。  
1、前一种AP1202GI的AP 提供的2.4G和5G 都能正常连接。但连接AP1202GI的AP时，2.4G相比5G更稳定 速率更快些。 
2、后一种AP1300GI的AP提供的2.4GWIFI可以连，但5GWIFI 连不上 。之前一直在AP1300GI所在房间里调试，所以一直没连上5G。 
不知道AP之间的差异是什么 ？