OpenIntelWireless/itlwm #714: panic(cpu 0 caller 0xffffff801e727998): "kfree: size 16148798281972502528 > kalloc_largest_allocated 96448512"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-6153.141.2/osfmk/kern/kalloc.c:868

aminobay, 2021-11-13:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.0.0 stable
- WiFi Card Model: Intel(R) Wi-Fi 6 AX200 160MHz
- PCI Product ID: 
- macOS Version: 10.15.7

**Description**
Freeze then Kernel panic, occurred once during 1.5hr continuous usage:

------------------------------------------------------------------------
panic(cpu 0 caller 0xffffff801e727998): "kfree: size 16148798281972502528 > kalloc_largest_allocated 96448512"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-6153.141.2/osfmk/kern/kalloc.c:868
Backtrace (CPU 0), Frame : Return Address
0xffffff83d51eb790 : 0xffffff801e71a65d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff83d51eb7e0 : 0xffffff801e854a75 mach_kernel : _kdp_i386_trap + 0x155
0xffffff83d51eb820 : 0xffffff801e8465fe mach_kernel : _kernel_trap + 0x4ee
0xffffff83d51eb870 : 0xffffff801e6c0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff83d51eb890 : 0xffffff801e719d27 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff83d51eb990 : 0xffffff801e71a117 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff83d51eb9e0 : 0xffffff801eec1a6c mach_kernel : _panic + 0x54
0xffffff83d51eba50 : 0xffffff801e727998 mach_kernel : _kfree + 0x1b8
0xffffff83d51ebaa0 : 0xffffff801edf1cb3 mach_kernel : _IOFree + 0x13
0xffffff83d51ebac0 : 0xffffff7fa3c893a7 com.zxystd.itlwm : __ZL4freePv + 0x47
0xffffff83d51ebaf0 : 0xffffff7fa3c8ff3a com.zxystd.itlwm : __ZN6ItlIwx12iwx_send_cmdEP9iwx_softcP12iwx_host_cmd + 0x6da
0xffffff83d51ebc00 : 0xffffff7fa3c974a9 com.zxystd.itlwm : __ZN6ItlIwx19iwx_send_cmd_statusEP9iwx_softcP12iwx_host_cmdPj + 0x49
0xffffff83d51ebc60 : 0xffffff7fa3c91ff0 com.zxystd.itlwm : __ZN6ItlIwx23iwx_send_cmd_pdu_statusEP9iwx_softcjtPKvPj + 0x90
0xffffff83d51ebcf0 : 0xffffff7fa3c91d1f com.zxystd.itlwm : __ZN6ItlIwx14iwx_sta_rx_aggEP9iwx_softcP14ieee80211_nodehttii + 0x19f
0xffffff83d51ebdc0 : 0xffffff7fa3c93025 com.zxystd.itlwm : __ZN6ItlIwx11iwx_ba_taskEPv + 0x335
0xffffff83d51ebe40 : 0xffffff7fa3d11a7f com.zxystd.itlwm : __Z12taskq_threadPv + 0x6f
0xffffff83d51ebfa0 : 0xffffff801e6c013e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.itlwm(2.0)[33B0CBAD-6F5A-3DB7-A01D-92CC39549103]@0xffffff7fa3c86000->0xffffff7fa4957fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f9f065000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[44472E6F-8DA0-3B46-ADEF-AFF76EC6C6DB]@0xffffff7f9f0ee000

BSD process name corresponding to current thread: kernel_task
Boot args: -v debug=0x100 keepsyms=1 alcid=1 npci=0x2000 

Mac OS version:
19H2

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x000000001e400000
Kernel text base: 0xffffff801e600000
__HIB  text base: 0xffffff801e500000
System model name: iMacPro1,1 (Mac-7BA5B2D9E42DDD94)
System shutdown begun: NO
Panic diags file available: NO (0xe00002bc)

System uptime in nanoseconds: 5052102865426

------------------------------------------------------------------------

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v2.0.0/itlwm_v2.0.0_stable.kext.zip
https://github.com/OpenIntelWireless/HeliPort/releases/download/v1.4.1/HeliPort.dmg

zxystd, 2021-11-14:
Is it reproducible? the panic seems impossible to happen, it means your system memory is corrupted by some reason, so this driver crash.

aminobay, 2021-11-14:
not reproduced, can a wild pointer cause this?

zxystd, 2021-11-23:
any further progress? I don't heard any panic like this one from others.

aminobay, 2021-11-23:
I was busy lately, Did not frequently use,   any test conditions you
recommend to reproduce the issue?

zxystd, 2024-03-16:
Resolved in v2.3.0-alpha.

closed 2024-03-16 by zxystd