OpenIntelWireless/itlwm #715: Internet doesn't work with DHCP or after assigning manual ip addresses

deepsidhu1313, 2021-11-22:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: 2.0.0
- WiFi Card Model: Intel 3160
- PCI Product ID: 0x08B3
- macOS Version: 11.6.1 (Big sur)

**Description**
I am unable to connect to the internet with 2.0.0 Kexts. I have tried both Airport and Itlwm. With DHCP system doesn't even assign ip from the DHCP range. I keep seeing exclamation sign. I also tried after assigning IP manually, with this method exclamation sign goes away and in network preferences, it says it is connected. But unable to ping any IP of local machines or remote systems. I am also unable to visit any website.

**Bug Report Archive**
[bugreport_62205.zip](https://github.com/OpenIntelWireless/itlwm/files/7578069/bugreport_62205.zip)

**Kext Download Source**
Downloaded Kext from From Github releases page

zxystd, 2021-11-22:
use v2.1.0-alpha version instead, if you are connecting wpa2/wpa3 mixed mode wifi, please use itlwm.kext

deepsidhu1313, 2021-11-23:
Thanks @zxystd , i can confirm using latest alpha release (e152cd5) of itlwm.kext, internet works as expected.

closed 2021-11-23 by deepsidhu1313