OpenIntelWireless/itlwm #716: after sleep wifi not work

Edwardwich, 2021-11-22:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 2.1alpha
- WiFi Card Model: n-7265
- PCI Product ID: 0x095a
- macOS Version: Mac OS 12.1beta2

**Description**
<after sleep wifi not work>

**Bug Report Archive**

 kextstat | grep -i "itlwm"
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   85    0 0                  0xf11000   0xf11000   com.zxystd.AirportItlwm (2.1.0) D1540328-950C-36B0-B4D7-CA53AA299A58 <84 49 16 7 6 3 1>

**Kext Download Source**

<https://github.com/OpenIntelWireless/itlwm/releases>

zxystd, 2021-11-23:
The log doesn't contains any useful information, you should collect the log ASAP the issue happened.

Edwardwich, 2021-11-23:
> The log doesn't contains any useful information, you should collect the log ASAP the issue happened.

How ?
Can you send terminal Command ؟

closed 2021-11-24 by Edwardwich