OpenIntelWireless/itlwm #720: Intel(R) Centrino(R) Wireless-N + WiMAX 6150 0x0886

jief666, 2021-12-04:
Hi.
I have a Intel(R) Centrino(R) Wireless-N + WiMAX 6150, pci id 8086:0886.
It looks like it's in your compatibility list, but not in the pci id list in the source.

Is it supported ?

Thanks.

zxystd, 2021-12-05:
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7655621/itlwm.kext.zip)
@jief666  Can you try this one?

jief666, 2021-12-05:
Thanks. It kind of works.
I have variable ping time to my router. Can go from 15 ms to 300ms. Same computer with another card was always 15ms.
I'm testing it since 2 minutes. Once, packets stopped going through although HeliPort was still reporting the connection as connected. I turned wifi off and on and it came back.

jief666, 2021-12-05:
Lost connection again, but it came back without me turning wifi off and on.

zxystd, 2021-12-05:
can you please follow our doc to collect running logs when it is disconnected?

jief666, 2021-12-05:
It's now running since 10 minutes. Ping time is still variable but mostly between 15 and 30ms. I've just thought trying Windows and ping time is also variable between 15-30ms.
I've just rebooted. Ping start go up to 800ms at first and seems to mostly decrease after few seconds.
I'm letting ping running to see if there is some break up.

jief666, 2021-12-05:
Works ok since 26 minutes now...
Nice work. Thanks.

zxystd, 2021-12-10:
@jief666 Can you help to test the latest v2.1.0-alpha version also works fine? Thank you.

jief666, 2021-12-18:
Hi. I'm sorry. I wish I could help.
I installed the driver you sent me on my friend's computer. Now I'm away with no access to the computer anymore.

zxystd, 2021-12-22:
@jief666 Ok, Thank you.

closed 2021-12-22 by zxystd