OpenIntelWireless/itlwm #723: itlwm.kext 2.0.0 with Killer 1650 not working over BigSur

lorenzocavallero, 2021-12-15:
**Have You Read Our Docs**
Yes ofc

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0 stable
- WiFi Card Model: Killer 1650
- PCI Product ID: idk
- macOS Version: 11.6.2

**Description**
Hi there, I just installed Big Sur on my DELL XPS 7590 and everything works fine so far except for wifi and bluetooth. I have the card I specified above and I loaded the kext from my OC config and then downloaded and installed Heliport but I still have no connection. I can't turn the wifi on from heliport and BT isn't working either. Any help would be really appreciated 

**Bug Report Archive**
[bugreport_15632.zip](https://github.com/OpenIntelWireless/itlwm/files/7721050/bugreport_15632.zip)

**Kext Download Source**
From github's release page

zxystd, 2021-12-16:
Have you inject some properties on PCI configuration?

lorenzocavallero, 2021-12-16:
I'm not quite sure about what you mean by "injecting properties on PCI configuration" , if you want i can just upload my EFI so you can check

williambj1, 2021-12-26:
@lorenzocavallero 

Please check if there are incompatible device properties injected in the bootloader's config, please remove any of them except the ones for graphics and audio.

Also please try the latest RC as well, some fixes for killer cards were added recently.

fsnhljz, 2021-12-27:
v2.1.0 RC1可以使用，非常好

closed 2022-01-04 by zxystd