OpenIntelWireless/itlwm #725: killer 1550i 160MHz版本的支持

TonyStark10006, 2021-12-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: AirportItlwm-Monterey-v2.1.0-DEBUG-alpha-2a84895
- WiFi Card Model: killer 1550i 160MHz
- PCI Product ID: 0x06F0
- macOS Version: 12.1 (21C52)

**Description**

老哥好。
网卡不工作，硬件信息如图一。
我看到itlwm/hal_iwx/ItlIwx.cpp里面有定义常量iwl9560_killer_1550i_160_name但是没有具体的实现代码，于是模仿ax201和9560 160MHz这两种实现方法添加代码，但是都轻松翻车。
我看文档也没有说支持160Mhz版本的1550i，试过Ubuntu 20.04.3 LTS，开箱也是蓝牙可用WiFi不能用的。请问目前固件是否支持这网卡，可行的能否指点一下实现思路？

致敬

图一
[image]

**Bug Report Archive**

抓不到日志

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/download/v2.1.0-alpha/AirportItlwm-Monterey-v2.1.0-DEBUG-alpha-2a84895.zip

zxystd, 2021-12-22:
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7761212/AirportItlwm.kext.zip)
看起来像是Linux的一个bug，可能Intel Linux团队的人也没有对这个卡进行过测试吧，itlwm本身来源于Linux的iwlwifi代码
你试试这个，日志一定是有的，也一定要，开机以后尽可能快的抓日志，不然可能会被系统的其他日志刷掉。没有日志的话没办法帮你。

TonyStark10006, 2021-12-22:
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7761212/AirportItlwm.kext.zip) 看起来像是Linux的一个bug，可能Intel Linux团队的人也没有对这个卡进行过测试吧，itlwm本身来源于Linux的iwlwifi代码 你试试这个，日志一定是有的，也一定要，开机以后尽可能快的抓日志，不然可能会被系统的其他日志刷掉。没有日志的话没办法帮你。

用上面的kext、最新α版的AirportItlwm.kext和itlwm.kext都是报这个错误
[image]

zxystd, 2021-12-23:
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7766181/AirportItlwm.kext.zip)
对的，有日志就好办了，再试试这个

TonyStark10006, 2021-12-23:
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7766181/AirportItlwm.kext.zip) 对的，有日志就好办了，再试试这个

你的kext可以，我这换了固件也可以了。
[image]

谢谢老哥了，致敬。

closed 2021-12-25 by TonyStark10006