OpenIntelWireless/itlwm #726: pci8086,6f0 ax201 在monterey下需要特别设置么? 总是无法驱动

steve3d, 2021-12-25:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: any version
- WiFi Card Model: Intel Ax201
- PCI Product ID: pci8086,6f0
- macOS Version: monterey 12.1

**Description**
我主板是z490i上自带的ax201, 记得在升级monterey之前, wifi是可以打开的, 但升级到monterey 之后, wifi就一直是无法驱动的状态.

```
kextstat |grep Air
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   73    0 0                  0xfa4000   0xfa4000   com.zxystd.AirportItlwm (2.1.0) AA2F647F-DBD4-3F52-A306-EEA1D89DCD21 <72 49 16 7 6 3 1>
```

驱动是正常加载了, 但就wifi图标上有个感叹号, 无法链接wifi

```
  名称：	pci8086,6f0
  类型：	Network controller
  已安装驱动程序：	否
  MSI：	是
  总线：	PCI
  插槽：	Built In
  供应商ID：	0x8086
  设备ID：	0x06f0
  子系统供应商ID：	0x8086
  子系统ID：	0x0074
  修正版ID：	0x0000
  链接宽度：	x0
```

pci里的信息..

zxystd, 2021-12-25:
2.0.0和最新的2.1.0都用过?之前能用是哪个版本?
日志一定要抓一下，实在抓不到的话开机-v用相机拍出来都行，没有日志不知道你这是什么情况。

steve3d, 2022-01-07:
```
Timestamp                       (process)[PID]    
2022-01-07 16:16:29.893104+0800  localhost kernel[0]: (IO80211FamilyLegacy) <IO80211FamilyLegacy`IO80211Controller::init(OSDictionary*)> AirportItlwm::init IO80211Legacy_kexts-30.1 "IO80211Legacy_kexts-30.1" Nov 30 2021 21:31:36
2022-01-07 16:16:38.397161+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> Sandbox: AirPlayXPCHelper(163) deny(1) mach-lookup com.apple.powerlog.plxpclogger.xpc
2022-01-07 16:16:48.161201+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> Sandbox: itunescloudd(672) deny(1) user-preference-write com.apple.airplay
2022-01-07 16:16:48.247466+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> 1 duplicate report for Sandbox: itunescloudd(672) deny(1) user-preference-write com.apple.airplay
2022-01-07 16:17:06.503374+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> Sandbox: cloudpaird(778) deny(1) mach-lookup com.apple.tccd
2022-01-07 16:17:07.783051+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> Sandbox: cloudpaird(778) deny(1) system-fsctl _IO('h', 47)
2022-01-07 16:17:07.835258+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> Sandbox: cloudpaird(778) deny(1) mach-lookup com.apple.metadata.mds
2022-01-07 16:17:09.561927+0800  localhost kernel[0]: (Sandbox) <Sandbox`log_kernel_report_summary> 1 duplicate report for Sandbox: cloudpaird(778) deny(1) mach-lookup com.apple.metadata.mds
```

日志里有Air开头的就这一行, 什么也没有, 之前是big sur能用, 但升级monterey后, 不管哪个版本都不能用,

zxystd, 2022-01-08:
11能用，12不能用那就不是驱动的问题了。

closed 2022-01-08 by zxystd