OpenIntelWireless/itlwm #727: AirportItlwm not working with Centrino Wireless-N 2230

mswgen, 2022-01-02:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: both 2.0.0 Stable and 2.1.0 Stable(Mojave version)
- WiFi Card Model: Intel Centrino Wireless-N 2230
- PCI Product ID: 8086:4062
- macOS Version: macOS Mojave 10.14.6(18G103)

**Description**
When I connect to Wi-Fi, it gets an ip but can't connect to internet. Pinging and ip(ex. 1.1.1.1) shows request timeout.
My laptop is HP Pavilion 15-e044TX.
Tried resetting preferences but didn't work.
Due to my SMBIOS(MacBookPro10,2) not supporting Secure Boot, I force loaded IO80211Family.

**Bug Report Archive**
[bugreport_27842.zip](https://github.com/OpenIntelWireless/itlwm/files/7798482/bugreport_27842.zip)

**Kext Download Source**
GitHub release page

mswgen, 2022-01-02:
Update: it works fine with my phone's personal hotspot. Maybe an issue with router.

williambj1, 2022-01-03:
Logs show the driver is working perfectly fine and the access point is properly connected.

- Have you tried `itlwm.kext`?
- Have you tried to connect to the same network in macOS Recovery?

mswgen, 2022-01-03:
> Logs show the driver is working perfectly fine and the access point is properly connected.
> 
> - Have you tried `itlwm.kext`?
> - Have you tried to connect to the same network in macOS Recovery?

- Havn't tried `itlwm.kext`.
- It works well in Recovery even without force loading IO80211Family.

closed 2022-01-05 by williambj1