OpenIntelWireless/itlwm #728: Airport Itlwm crashes with private relay😥

singhalrishi27, 2022-01-03:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Not Sure

**Environment**

- Kext Version: Airport itlwm  2.1.0 Stable
- WiFi Card Model: 0x8086 (intel 7260)
- PCI Product ID: 0x4470 (intel 7260)
- macOS Version: 

**Description**
When using iCloud Private Relay and doing usual work causes macOS to Crash that's why I wasn't able to generate the report.
Most likely you will face the issue of you enable private relay...🥲🥲🥲

And no when private relay is off everything just works...
**Bug Report Archive**
Not Included coz Computer crashes if you want I will include it

**Steps To Reproduce**

It not exact that macOS will crash sometime wifi menu will not respond sometime you nothing will respond Sometimes everything will work just fine
Connect your wifi 
Turn on private relay 
Browse safari and see the magic✨

**Kext Download Source**
Release section of your Repository

**Device**
Redmi note 7s
Evolution X custom rom based on  Android 12 
11ac wifi mode
WiFi secured with WPA2 Personal passcode
No network is not hidden

williambj1, 2022-01-04:
@singhalrishi27 

- Does it trigger a kernel panic? If so please attach the panic logs.
- Does it crash if itlwm.kext is used instead?
- Have you tried to reset the network interfaces? (instructions can be found in the docs)

Thanks.

Unfortunately I didn’t purchase this service and it is unavailable in the county zxystd lives in so it’s not very convenient fir us to verify. If other are facing the same issue feel free to share them below.

singhalrishi27, 2022-01-04:
> @singhalrishi27
> 
> * Does it trigger a kernel panic? If so please attach the panic logs.
> * Does it crash if itlwm.kext is used instead?
> * Have you tried to reset the network interfaces? (instructions can be found in the docs)
> 
> Thanks.
> 
> Unfortunately I didn’t purchase this service and it is unavailable in the county zxystd lives in so it’s not very convenient for us to verify. If others are facing the same issue feel free to share them below.

panic(cpu 0 caller 0xffffff802b2bc7cf): mbuf_watchdog: 1 waiters stuck for 21 secs
28635/28665 mbufs in use:
	28635 mbufs allocated to data
	30 mbufs allocated to caches
636/682 mbuf 2KB clusters in use
13521/13521 mbuf 4KB clusters in use
0/0 mbuf 16KB clusters in use
62680 KB allocated to network (approx. 99% in use)
0 KB returned to the system
VM allocation failures: contiguous 0, normal 0, one page 0
worker thread runs: 6257, expansions: 97, cl 0/0, bigcl 6256/123707, 16k 0/0
worker thread last run time: 1564 (22 seconds ago)

mbuf leak detection table:
	total captured: 936 (one per 500)
	total allocs outstanding: 93
	new hash recorded: 902 allocs, 686 traces
	hash collisions: 22 allocs, 27 traces
	overwrites: 496 allocs, 0 traces
	lock conflicts: 0

top 5 outstanding traces:
[1] 25 outstanding alloc(s), 25 hit(s), 0 collision(s)
[2] 13 outstanding alloc(s), 24 hit(s), 0 collision(s)
[3] 13 outstanding alloc(s), 161 hit(s), 0 collision(s)
[4] 12 outstanding alloc(s), 27 hit(s), 0 collision(s)
[5] 12 outstanding alloc(s), 18 hit(s), 0 collision(s)

    trace [1]           trace [2]           trace [3]       	trace [4]           trace [5]      
    ------------------  ------------------  ------------------      ------------------  ------------------ 
 1: 0xffffff80008baf7a  0xffffff80008baf7a  0xffffff80008baf7a  0xffffff80008baf7a  0xffffff80008baf7a  
 2: 0xffffff80008948b2  0xffffff80008948b2  0xffffff8000894810  0xffffff8000894810  0xffffff80008948b2  
 3: 0xffffff80008b9e49  0xffffff80008bf1e8  0xffffff80008bf1e8  0xffffff80008b9df6  0xffffff80008c17f2  
 4: 0xffffff800089477f  0xffffff80008c3491  0xffffff80008c3491  0xffffff800089477f  0xffffff800088eae1  
 5: 0xffffff80008c17f2  0xffffff800088dea3  0xffffff800088dea3  0xffffff80008c17f2  0xffffff8002b4506d  
 6: 0xffffff800088eae1  0x0  0x0  0xffffff800088eae1  0x0  
 7: 0xffffff8002b4506d  0x0  0x0  0xffffff8002b4506d  0x0  
 8: 0x0  0x0  0x0  0x0  0x0  
 9: 0x0  0x0  0x0  0x0  0x0  
10: 0x0  0x0  0x0  0x0  0xffffff8000a58d13  
11: 0x0  0x0  0x0  0x0  0xffffff8000a5756e  
12: 0xffffff8000a58d13  0x0  0x0  0xffffff8000a58d13  0xffffff8000a56b97  
13: 0xffffff8000a5756e  0x0  0x0  0xffffff8000a5756e                      
14: 0xffffff8000a56b97  0x0  0x0  0xffffff8000a56b97                      
15:                     0x0  0x0                                          
16:                     0xffffff8000a58d13  0xffffff8000a58d13                                          

Panicked task 0xffffff9510d623b0: 4 threads: pid 195: mDNSResponder
Backtrace (CPU 0), panicked thread: 0xffffff8b7b77c540, Frame : Return Address
0xffffffdc730fb650 : 0xffffff802ac87ddd 
0xffffffdc730fb6a0 : 0xffffff802ade7d15 
0xffffffdc730fb6e0 : 0xffffff802add74e3 
0xffffffdc730fb730 : 0xffffff802ac27a60 
0xffffffdc730fb750 : 0xffffff802ac881ad 
0xffffffdc730fb870 : 0xffffff802ac87966 
0xffffffdc730fb8d0 : 0xffffff802b5160b9 
0xffffffdc730fb940 : 0xffffff802b2bc7cf 
0xffffffdc730fb9d0 : 0xffffff802b2ba502 
0xffffffdc730fba30 : 0xffffff802b29477f 
0xffffffdc730fbaa0 : 0xffffff802b2b9e49 
[... 183 lines cut ...]
|IOUSBMass!SDriver	209.40.6
|IOSCSIBlockCommandsDevice	452.60.2
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	452.60.2
>!AMobileFileIntegrity	1.0.5
$!AImage4	4.2.0
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.Libm	1
@kec.corecrypto	12.0

singhalrishi27, 2022-01-04:
> Thanks.

2.)nope not tested with heliport
3.)yes I even reinstalled macOS

zxystd, 2022-01-05:
@singhalrishi27 Hi, thank you for reporting this issue, can you please add `keepsyms=1` to `boot-args` then reproduce the issue again? I think it is an airport related issue.

singhalrishi27, 2022-01-05:
> @singhalrishi27 Hi, thank you for reporting this issue, can you please add `keepsyms=1` to `boot-args` then reproduce the issue again? I think it is an airport related issue.

Do I have to use debug kexts?

zxystd, 2022-01-05:
@singhalrishi27 yes, debug kext is needed. panic logs, kext logs would be well if possible to get that.

williambj1, 2022-01-05:
All alpha version kexts are built with debug features enabled.

singhalrishi27, 2022-01-05:
@zxystd Hey I was able to collect a lot of logs for you.
**What I did -** 
installed latest alpha kext
added Debug Enhancer.kext
added keepsyms=1 to boot-args
Took Clean Snapshot of config.plist
and restarted macOS
after that, I started collecting logs when everything was working after some time itlwm crashed

https://user-images.githubusercontent.com/67616808/148171504-1461b534-cafa-45a3-8686-a59aec5fa695.mov

in this clip, you can see that I can't turn off wifi    
also, network preferences is empty 
after a while macos crashes 
from crash I mean macOS stuck at just one screen no inputs works like trackpad will do nothing you can't move your cursor etc you have to force restart your Laptop I'm attaching a log in a separate comment...
@williambj1

singhalrishi27, 2022-01-05:
[Panic_LOG_After_ITLWM_crash.txt](https://github.com/OpenIntelWireless/itlwm/files/7812528/Panic_LOG_After_ITLWM_crash.txt)
[Log_2022-01-05_11-41-39.log](https://github.com/OpenIntelWireless/itlwm/files/7812529/Log_2022-01-05_11-41-39.log)
[Log_2022-01-05_11-40-18.log](https://github.com/OpenIntelWireless/itlwm/files/7812530/Log_2022-01-05_11-40-18.log)
[Log_2022-01-05_11-39-35.log](https://github.com/OpenIntelWireless/itlwm/files/7812531/Log_2022-01-05_11-39-35.log)
[Log_2022-01-05_11-38-34.log](https://github.com/OpenIntelWireless/itlwm/files/7812532/Log_2022-01-05_11-38-34.log)
[Log_2022-01-05_11-38-17.log](https://github.com/OpenIntelWireless/itlwm/files/7812533/Log_2022-01-05_11-38-17.log)

all logs i collected

Overc1ocker, 2022-01-08:
Immediate when the issue starts, put your mac to sleep (wait for the indicator light to show it's sleeping), then wake it back up. Attach the logs after the wake. Did it fix the wifi?

singhalrishi27, 2022-01-08:
> Immediately when the issue starts, put your mac to sleep (wait for the indicator light to show it's sleeping), then wake it back up. Attach the logs after the wake. Did it fix the wifi?

Nope it doesn't after wake system becomes unresponsive after a while and you have to force restart it after that same log message appears and asks me to send report to apple

zxystd, 2022-01-09:
> [Panic_LOG_After_macOS_Restart.txt](https://github.com/OpenIntelWireless/itlwm/files/7812528/Panic_LOG_After_ITLWM_crash.txt) [Log_2022-01-05_11-41-39.log](https://github.com/OpenIntelWireless/itlwm/files/7812529/Log_2022-01-05_11-41-39.log) [Log_2022-01-05_11-40-18.log](https://github.com/OpenIntelWireless/itlwm/files/7812530/Log_2022-01-05_11-40-18.log) [Log_2022-01-05_11-39-35.log](https://github.com/OpenIntelWireless/itlwm/files/7812531/Log_2022-01-05_11-39-35.log) [Log_2022-01-05_11-38-34.log](https://github.com/OpenIntelWireless/itlwm/files/7812532/Log_2022-01-05_11-38-34.log) [Log_2022-01-05_11-38-17.log](https://github.com/OpenIntelWireless/itlwm/files/7812533/Log_2022-01-05_11-38-17.log)
> 
> all logs i collected

Got it, thank you. But I don't have any clue on how it happened right now...

singhalrishi27, 2022-01-09:
> Got it, thank you. But I don't have any clue on how it happened right now...

Any other way we can fix it?

williambj1, 2022-01-10:
> Any other way we can fix it?

Consider trying itlwm.kext with HeliPort?

zxystd, 2022-01-13:
> Any other way we can fix it?

@singhalrishi27 Still doing research, but from what I received the similar issues have the common thing that they all using macOS 12, so can you help me tested on BigSur or Catalina? or can you use Heliport for test? 
Thanks a lot.

singhalrishi27, 2022-01-19:
> > Any other way we can fix it?
> 
> @singhalrishi27 Still doing research, but from what I received the similar issues have the common thing that they all using macOS 12, so can you help me tested on BigSur or Catalina? or can you use Heliport for test? Thanks a lot.

@williambj1 
@zxystd 
i tried with heliport and issue still remains.
After a while Heliport will crash in activity moniter heliport will not respond.
if you force quit heliport 
it will take 30 seconds just to foce quit the app after force quitting Helipot macOS will crash

williambj1, 2022-01-19:
@singhalrishi27 Interesting results and thank you for testing. Does Ethernet work fine with the private relay?

singhalrishi27, 2022-01-19:
> @singhalrishi27 Interesting results and thank you for testing. Does Ethernet work fine with the private relay?

I don't have ethernet in india we mostly use cellular 😅 btw change the tag from question to a bug

williambj1, 2022-01-19:
> > @singhalrishi27 Interesting results and thank you for testing. Does Ethernet work fine with the private relay?
> 
> I don't have ethernet in india we mostly use cellular 😅 btw change the tag from question to a bug

At this point it still unclear whether there is a bug in Safari/private relay shipped in macOS Monterey or the issue is directly related to itlwm. Theoretically, user space software (in this case, a browser + a proxy) should not be able to trigger kernel level crashes if another other ways of browsing the internet works fine.

We have quite a lot of people using Monterey in our internal testing group and so far I haven't heard of any similar crashes; a few people are using the same wifi card and version of os as yours. However we've recently got in touch with the author of #735 which also have a very similar crash but that person uses a different wifi card.

singhalrishi27, 2022-01-19:
> > > @singhalrishi27 Interesting results and thank you for testing. Does Ethernet work fine with the private relay?
> > 
> > 
> > I don't have ethernet in india we mostly use cellular 😅 btw change the tag from question to a bug
> 
> At this point it still unclear whether there is a bug in Safari/private relay shipped in macOS Monterey or the issue is directly related to itlwm. Theoretically, user space software (in this case, a browser + a proxy) should not be able to trigger kernel level crashes if another other ways of browsing the internet works fine.
> 
> We have quite a lot of people using Monterey in our internal testing group and so far I haven't heard of any similar crashes; a few people are using the same wifi card and version of os as yours. However we've recently got in touch with the author of #735 which also have a very similar crash but that person uses a different wifi card.
> 
> Test results from other network adapters will help a lot. If Ethernet is not available, sharing the network from a phone can also do the trick.

@williambj1 
You are right first i thought private relay is in beta so might not be stable but likely impossible a service can occur kernel panic i will try with ethernet then I will tell you what happens.
My Uncle  has a real mac he uses private relay and it works just as well as not using private relay. Hope i somehow manages to get hands on with someone else's ethernet and will try to produce the issue

singhalrishi27, 2022-03-04:
@williambj1 @zxystd i tested with ethernet private relay works surprisingly well with it. I can confirm that the issue is indeed with airportitlwm and itlwm🥲

zxystd, 2022-03-25:
can you check if the v2.2.0-alpha have the same panic? I don't see anyone reported the same panic yet.

singhalrishi27, 2022-03-25:
> can you check if the v2.2.0-alpha have the same panic? I don't see anyone reported the same panic yet.

Yup issue still exists

zxystd, 2022-03-27:
@singhalrishi27 O, can you also try connecting to an open wifi which means no encryptions to see if the issue still persist? Or try to use another router if you can.

singhalrishi27, 2022-05-31:
> @singhalrishi27 O, can you also try connecting to an open wifi which means no encryptions to see if the issue still persist? Or try to use another router if you can.

Issue still persists

chowaniec, 2022-09-05:
I have the exact same issue with private relay running the latest 2.2 alpha kext on Monterey. Lenovo T480 Intel 8265 card. With private relay off - everything works rock solid.