OpenIntelWireless/itlwm #730: While on Safari, hackintosh frooze. + Wi-fi disconnects from hour to hour.

plae2084, 2022-01-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: The latest one (v2.1.0)
- WiFi Card Model: Intel AX200 Wi-Fi 6
- PCI Product ID: 0x2723
- macOS Version: 12.1 (macOS Monterey)

**Description**
Okay, so. I was normally listening to some music on Safari (youtube). I think I was listening for 5 hours straight. Until I wanted to pause the video, the system crashed. In the panic log, there is something about the intel kext so that's why I am making this bug report. Maybe I should try to listen on Spotify but meeh.
The second issue is the Wi-Fi randomnly disconnects in a certain period of time. When it is connected for straight one hour, it disconnects and then reconnects succesfully. The thing is that sometimes I am playing online games and it's annoying to disconnects in the gameplay. In my opninon, I think I should replace the card but It'll deliver so slow. Anyway, I know that this project is still in development so that's alright.

**Bug Report Archive**
[Log_2022-01-08_21-31-04.log](https://github.com/OpenIntelWireless/itlwm/files/7833877/Log_2022-01-08_21-31-04.log)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.1.0

williambj1, 2022-01-09:
Hi, please upload the panic log. Try to increase the DHCP lease time in your router and see if it helps.

plae2084, 2022-01-09:
> Hi, please upload the panic log. Try to increase the DHCP lease time in your router and see if it helps.

Ugh, that means I need to redo the problem. I am coming back when I get to the same result. Also, how do I increase the DHCP lease time in my router?

zxystd, 2022-01-09:
@plae2084 can you add `keepsyms=1` at `boot-args` and then try again? or it will not contain the actual symbol and hard to know what's going wrong. 

>Also, how do I increase the DHCP lease time in my router?

Correct that the option is key exchange period,  It depends on your router, most likely that the default value is 60 minutes.

Please not that the panic info also save on you Mac when kernel panic happens, you can find the files at /Library/Logs/DiagnosticReports/ which ends with `.panic`.

plae2084, 2022-01-09:
> @plae2084 can you add `keepsyms=1` at `boot-args` and then try again? or it will not contain the actual symbol and hard to know what's going wrong

Added it. What should I do now? Should I make the same thing to make the system crash and get the log?

> Correct that the option is key exchange period, It depends on your router, most likely that the default value is 60 minutes.

Okay, I'll try and I'll let you know.

> Please not that the panic info also save on you Mac when kernel panic happens, you can find the files at /Library/Logs/DiagnosticReports/ which ends with `.panic`.

That's what I did. I found it there. But I copied it into a txt file cause GitHub doesn't let me upload .panic type files.

zxystd, 2022-01-09:
> Added it. What should I do now? Should I make the same thing to make the system crash and get the log?
> 
Reboot and reproduce it again, thank you.

zxystd, 2022-01-09:
Another question, have you ever used v2.0.0? it is the same?

plae2084, 2022-01-09:
> Another question, have you ever used v2.0.0? it is the same?

Nah. I just got the card and I tried the latest version. I had Big Sur installed 2 days ago and the network was running fine I guess? I mean I haven't did much thing on Safari then. Gotta try it and let you know. I am still trying to make the system crash again also.

plae2084, 2022-01-10:
> Nah. I just got the card and I tried the latest version. I had Big Sur installed 2 days ago and the network was running fine I guess? I mean I haven't did much thing on Safari then. Gotta try it and [...]

Sorry for the long response. Using v2.0.0 resolves the Wi-Fi disconnecting issue and also I've seen some improvements. I still don't know if it'll crash on Safari. Guess I gotta use Chrome. Thank you!

williambj1, 2022-01-15:
> Sorry for the long response. Using v2.0.0 resolves the Wi-Fi disconnecting issue and also I've seen some improvements. I still don't know if it'll crash on Safari. Guess I gotta use Chrome. Thank you!

@plae2084 Thank you for the reply. I would like to know a bit more information to help us clarify and generalize the issue.

Did the panic trigger again?
Have you tried this kext with other versions of macOS before?
Do the conditions in #728 apply to your setup as well?

stefan89, 2022-01-20:
I am facing the same issue on MacOS 11.6.2 using itlwm version 2.1.0

A downgrade to 2.0.0 is a workaround for now?

richibrics, 2022-01-21:
Hi @stefan89 can you check in your System Console if when it disconnects, you have EAPOL messages and/or a log like [this](https://github.com/OpenIntelWireless/itlwm/issues/680#issuecomment-924812996) ? Thank you

stefan89, 2022-01-23:
@richibrics, no I don't get those messages when WIFI disconnects and reconnects.

PS. downgrading to version 2.0.0 did not help for me.

zxystd, 2024-03-16:
inactive.

closed 2024-03-16 by zxystd