OpenIntelWireless/itlwm #731: 黑苹果网卡不定时断开重连

Jokerhoho, 2022-01-10:
一、系统版本：
```
    macOS big Sur 11.2
```
二、硬件信息:
```
    Hardware Overview:
      Model Name: iMac
      Model Identifier: iMac20,1
      Processor Name: 8-Core Intel Core i7
      Processor Speed: 3.8 GHz
      Number of Processors: 1
      Total Number of Cores: 8
      L2 Cache (per Core): 256 KB
      L3 Cache: 16 MB
      Hyper-Threading Technology: Enabled
      Memory: 64 GB
```

三、网卡型号：
```
    Intel AX201
```

四、故障说明：
```
    1、Wi-Fi不定时断开重连，偶尔重连不上去，就只能重启链接，测试驱动版本1.2.0以上存在此问题。
    2、但是如果使用1.2.0驱动版本，网速只有50m，无法满速加载。
    3、此外：路由器已经将密钥重制周期设置为0，即不交换。
```
五、错误日志：
[Uploading Log_2022-01-10_11-20-52.log…]()

williambj1, 2022-01-10:
@Jokerhoho 日志没有上传成功，麻烦重新上传，谢谢

Jokerhoho, 2022-01-10:
[Log_2022-01-10_11-51-45.log](https://github.com/OpenIntelWireless/itlwm/files/7836789/Log_2022-01-10_11-51-45.log)

Jokerhoho, 2022-01-10:
> @Jokerhoho 日志没有上传成功，麻烦重新上传，谢谢

已上传，麻烦了，谢谢！

zxystd, 2022-01-10:
[image]
@Jokerhoho  仍然发生了密钥重传，你的路由器那个设置并没有效果，建议换个路由器，或者换成itlwm+heliport先用着吧。至于为什么airport重传会失败我也不清楚，手里也没有这样的问题路由器没法测试

Jokerhoho, 2022-01-10:
> [image]
> 
> @Jokerhoho 仍然发生了密钥重传，你的路由器那个设置并没有效果，建议换个路由器，或者换成itlwm+heliport先用着吧。至于为什么airport重传会失败我也不清楚，手里也没有这样的问题路由器没法测试

好吧，那我测试下，我是网件的R7000刷的梅林系统。
[image]

zxystd, 2022-01-10:
[image]
那就没错了，GTK发生了重传，由路由器发起的，也就是Group Key，驱动更新Group Key之后就开始出错了，emmm，我研究下看看，如果你能配合测试下那就更好了。实在不行也可以关闭RSN_Supplicant让你也可以用上airportitlwm

Jokerhoho, 2022-01-10:
而且还有一个奇怪的问题是，可能刚过连接上过了一会就断了，有的时候时间很长才断。主板开启了cnvi，我可以帮忙测试下，我邮箱zinuo2019@gmail.com

Jokerhoho, 2022-01-10:
> 没错了，GTK发生了重传，由路由器发起的，也就是Group Key，驱动更新Group Key之后就开始出错了，emmm，我研究下看看，如果你能配合测试下那就更好了。实在不行也可以关闭RSN_Supplicant让你也可以用上airportitlwm

+qq:2358867058,我可以测试下

zxystd, 2022-01-10:
QQ群：1132224334
我已经大概发现问题了。。。

Jokerhoho, 2022-01-10:
> QQ群：1132224334 我已经大概发现问题了。。。

OK，已申请加群。

MHL3060, 2022-01-13:
I am experiencing the same issue after waking up from sleep. The latest snapshot works reliably on the first boot. After the wake up from sleep. the DHCP seems not able to get the IP address. (it gets an IP address and lost it repeatly)

zxystd, 2022-01-13:
collecting the running log if the symptom you have described happened. Thank you.

MHL3060, 2022-01-17:
here is the log when getting disconnect and connect repeatly
```
[265420.716584]: itlwm: itlwm: sending addba_req to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265420.716599]: itlwm: iwm_tx_fill_cmd ridx=7
[265420.717517]: itlwm: taskq task_add ba_task add to queue
[265420.717527]: itlwm: iwm_ba_task start=1 ssn=0, tid=0 scd_queue=10
[265420.717535]: itlwm: iwm_enable_txq qid=10 tid=0 agg=1
[265420.717782]: itlwm: iwm_add_sta_cmd line=319
[265420.717904]: itlwm: iwm_sta_tx_agg tx agg done. err=0 status=1, mask_status=1
[265420.751997]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265420.767875]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265420.811258]: itlwm: itlwm: dumping device error log
[265420.811408]: itlwm: itlwm: Start Error Log Dump:
[265420.811409]: itlwm: itlwm: Status: 0x39, count: 6
[265420.811410]: itlwm: itlwm: 0x00001030 | ADVANCED_SYSASSERT          
[265420.811412]: itlwm: itlwm: 00800222 | trm_hw_status0
[265420.811413]: itlwm: itlwm: 00000000 | trm_hw_status1
[265420.811414]: itlwm: itlwm: 00000B30 | branchlink2
[265420.811415]: itlwm: itlwm: 000164C0 | interruptlink1
[265420.811421]: itlwm: itlwm: 00000000 | interruptlink2
[265420.811422]: itlwm: itlwm: DEADBEEF | data1
[265420.811423]: itlwm: itlwm: DEADBEEF | data2
[265420.811424]: itlwm: itlwm: DEADBEEF | data3
[265420.811425]: itlwm: itlwm: FFFB3777 | beacon time
[265420.811426]: itlwm: itlwm: F36B9855 | tsf low
[265420.811427]: itlwm: itlwm: 00000017 | tsf hi
[265420.811428]: itlwm: itlwm: 00000B2C | time gp1
[265420.811428]: itlwm: itlwm: 0042EE24 | time gp2
[265420.811429]: itlwm: itlwm: 00000000 | uCode revision type
[265420.811430]: itlwm: itlwm: 00000011 | uCode version major
[265420.811431]: itlwm: itlwm: BFB58538 | uCode version minor
[265420.811432]: itlwm: itlwm: 00000144 | hw version
[265420.811433]: itlwm: itlwm: 00089004 | board version
[265420.811434]: itlwm: itlwm: 0A02001C | hcmd
[265420.811435]: itlwm: itlwm: A6F23002 | isr0
[265420.811436]: itlwm: itlwm: 01C18000 | isr1
[265420.811437]: itlwm: itlwm: 00000112 | isr2
[265420.811438]: itlwm: itlwm: 00417CC8 | isr3
[265420.811439]: itlwm: itlwm: 00000000 | isr4
[265420.811439]: itlwm: itlwm: 10004112 | last cmd Id
[265420.811440]: itlwm: itlwm: 00000000 | wait_event
[265420.811441]: itlwm: itlwm: 000000C4 | l2p_control
[265420.811442]: itlwm: itlwm: 0001D410 | l2p_duration
[265420.811443]: itlwm: itlwm: 00000007 | l2p_mhvalid
[265420.811444]: itlwm: itlwm: 00000081 | l2p_addr_match
[265420.811445]: itlwm: itlwm: 00000015 | lmpm_pmg_sel
[265420.811446]: itlwm: itlwm: 15041745 | timestamp
[265420.811447]: itlwm: itlwm: 0000E8F0 | flow_handler
[265420.811448]: itlwm: driver status:
[265420.811449]: itlwm:   tx ring  0: qid=0  cur=59  queued=0  
[265420.811450]: itlwm:   tx ring  1: qid=1  cur=0   queued=0  
[265420.811452]: itlwm:   tx ring  2: qid=2  cur=0   queued=0  
[265420.811453]: itlwm:   tx ring  3: qid=3  cur=0   queued=0  
[265420.811454]: itlwm:   tx ring  4: qid=4  cur=0   queued=0  
[265420.811468]: itlwm:   tx ring  5: qid=5  cur=95  queued=0  
[265420.811469]: itlwm:   tx ring  6: qid=6  cur=0   queued=0  
[265420.811470]: itlwm:   tx ring  7: qid=7  cur=0   queued=0  
[265420.811471]: itlwm:   tx ring  8: qid=8  cur=0   queued=0  
[265420.811472]: itlwm:   tx ring  9: qid=9  cur=0   queued=0  
[265420.811473]: itlwm:   tx ring 10: qid=10 cur=4   queued=2  
[265420.811475]: itlwm:   tx ring 11: qid=11 cur=0   queued=0  
[265420.811476]: itlwm:   tx ring 12: qid=12 cur=0   queued=0  
[265420.811477]: itlwm:   tx ring 13: qid=13 cur=0   queued=0  
[265420.811478]: itlwm:   tx ring 14: qid=14 cur=0   queued=0  
[265420.811479]: itlwm:   tx ring 15: qid=15 cur=0   queued=0  
[265420.811480]: itlwm:   tx ring 16: qid=16 cur=0   queued=0  
[265420.811481]: itlwm:   tx ring 17: qid=17 cur=0   queued=0  
[265420.811483]: itlwm:   tx ring 18: qid=18 cur=0   queued=0  
[265420.811484]: itlwm:   tx ring 19: qid=19 cur=0   queued=0  
[265420.811485]: itlwm:   tx ring 20: qid=20 cur=0   queued=0  
[... 197 lines cut ...]
[265424.433502]: itlwm: ieee80211_send_eapol_key
[265424.433503]: itlwm: ieee80211_eapol_key_mic
[265424.433544]: itlwm: ieee80211_must_update_group_key
[265424.433545]: itlwm: ieee80211_recv_4way_msg3 k_len=16 rsc=7B 29 04 00 00 00itlwm: ieee80211_set_key kid=1 cipher=8, error=0
[265424.433553]: itlwm: ieee80211_set_link_state nstate=4, old_state=2
[265424.433554]: itlwm: ieee80211_set_link_state LINK_STATE_IS_UP
[265424.435515]: Reinit'd ND information for interface en1
[265424.435556]: en1: starting optimistic DAD immediately for fe80:5::1c52:cf53:4ce:3e7e
[265424.435561]: itlwm: setMulticastList multicast count=5 bssid=9c:1e:95:a7:12:46
[265424.438227]: itlwm: taskq task_add ba_task add to queue
[265424.438306]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265424.438317]: itlwm: iwm_tx_fill_cmd ridx=7
[265424.440302]: itlwm: setMulticastList multicast count=6 bssid=9c:1e:95:a7:12:46
[265424.453263]: itlwm: taskq task_add ba_task add to queue
[... cut ...]
[265424.863842]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265424.909845]: itlwm: taskq task_add ba_task add to queue
[265424.910331]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265424.910342]: itlwm: iwm_tx_fill_cmd ridx=7
[265425.013812]: tx_flush:1182: disk3 xid 5275918 tx stats: # 201420 finish 201535 enter 95 wait 1376 1862us close 2us flush 1901us
```

williambj1, 2022-01-18:
@MHL3060 Please create a new issue, your card uses the iwm driver and the disconnection is caused by a firmware crash.

zxystd, 2022-04-02:
@Jokerhoho 帮忙试一下最新的alpha版本解决了这个问题了没

closed 2023-06-03 by zxystd