OpenIntelWireless/itlwm #734: Request: update documentation to disable "fast startup" in Windows in case the user has a dual boot system (Windows + MacOS)

stefan89, 2022-01-11:
After hours of research, I accidentally found the reason why my Intel AX200 Wi-Fi did not work with the itlwm.kext and AirportItlwm.kext (tried both individually). It is because I had "fast startup" enabled in Windows. When you have a dual boot system (Windows & MacOS) and this feature is enabled in Windows, it will prevent that you can use your Wi-Fi adapter in MacOS.

After disabling this feature in Windows, I finally was able to see the Wi-Fi adapter in MacOS and was able to connect.

I think I am not the only one who faces/faced this issue, so probably it is an idea to add this to the documentation so people are aware of this.

How to turn off fast startup in Windows can be found here:
https://www.computerhope.com/issues/ch001762.htm

stefan89, 2022-01-11:
Update: now I see there is a link to the solution in the documentation. But I think it could be more prominent visible in the documentation I think. Most people won't click on it when they have issues to get this kext up and running.
https://openintelwireless.github.io/itlwm/FAQ.html#dual-boot-with-windows

williambj1, 2022-01-11:
@stefan89 I've moved the info to the troubleshooting section, please let me know if It is better. Thank you for your suggestion.

closed 2022-01-11 by williambj1

stefan89, 2022-01-11:
@williambj1 , well I see it is now part of the Bios configuration subsection (https://openintelwireless.github.io/itlwm/Troubleshooting.html#bios-configuration) but that is not correct I think since it is not a BIOS setting. Fast startup is a Windows setting.

See: https://www.computerhope.com/issues/ch001762.htm

stefan89, 2022-01-11:
Maybe it is an idea to create a new subsection in the TroubleShooting section called
"Windows configuration (dual boot with MacOS)" or something like that

alyxferrari, 2022-01-11:
Windows must modify the BIOS setting for it somehow when you change it in Windows. It's definitely a BIOS setting. If it was truly a Windows setting, it would have no impact on macOS.