OpenIntelWireless/itlwm #735: 掉网卡，机器自动重启

SnowsonZ, 2022-01-12:
**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: Wireless 8265/8275
- PCI Product ID: 0x8086 0x24FD
- macOS Version: 12.0.1

**Description**
 看网页视频时频繁出现掉网卡，机器自动重启。必现。

**Bug Report Archive**

**Log**
[Log_2022-01-12_18-34-59.log](https://github.com/OpenIntelWireless/itlwm/files/7853993/Log_2022-01-12_18-34-59.log)

[Log_2022-01-12_18-35-23.log](https://github.com/OpenIntelWireless/itlwm/files/7853994/Log_2022-01-12_18-35-23.log)

[image]

zxystd, 2022-01-12:
机器自动重启以后弹框的内容呢？

SnowsonZ, 2022-01-12:
> 机器自动重启以后弹框的内容呢？

你是说重启后提示发送给apple的log？

SnowsonZ, 2022-01-12:
> 是的
[system.log](https://github.com/OpenIntelWireless/itlwm/files/7855875/system.log)

SnowsonZ, 2022-01-12:
> 是的
看清晰度高(2k,4k)的视频更容易发生这个问题。

zxystd, 2022-01-12:
哪个版本开始出现的这个问题，可以试一下2.0.0和1.3.0，看看是不是都有这个问题?

SnowsonZ, 2022-01-13:
> 哪个版本开始出现的这个问题，可以试一下2.0.0和1.3.0，看看是不是都有这个问题?

2.0.0版本试过也有这个问题，1.3好像是没事。我再试一下。

SnowsonZ, 2022-01-13:
> 哪个版本开始出现的这个问题，可以试一下2.0.0和1.3.0，看看是不是都有这个问题?

1.3没有Monterey版本

zxystd, 2022-01-13:
@SnowsonZ 用itlwm.kext + Heliport试试吧，我让人测试了8k 4k视频播放都没有问题

williambj1, 2022-01-15:
@SnowsonZ 方便进群测试吗 https://github.com/OpenIntelWireless/itlwm/issues/731#issuecomment-1008572120

SnowsonZ, 2022-01-15:
> @SnowsonZ 方便进群测试吗 [#731 (comment)](https://github.com/OpenIntelWireless/itlwm/issues/731#issuecomment-1008572120)

已申请

SnowsonZ, 2022-01-15:
> @SnowsonZ 方便进群测试吗 https://github.com/OpenIntelWireless/itlwm/issues/731#issuecomment-1008572120

又测了下，用chrome不会出现这个问题。safari必现

williambj1, 2022-01-19:
duplicate of #730

closed 2022-01-19 by williambj1

singhalrishi27, 2022-01-19:
You had private relay turned on?

778280151, 2022-03-27:
+1 同样的问题

zhuofan-16, 2023-04-23:
+1 Same Issue Encountered, I am also using 8265