OpenIntelWireless/itlwm #736: GIT 拉取项目时WiFi掉网

oxygen-02, 2022-01-17:
**Have You Read Our Docs**

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version:   AirportItlwm 2.1.0
- WiFi Card Model:  Intel AC8265
- PCI Product ID: 供应商id/0x8086、设备id/0x24FD
- macOS Version:  Big Sur 11.2

**Description**
1. 日常打开idea，然后通过idea拉取项目
2. WiFi连接断开然后再次连上
3. 目前只有mac自带控制台的wifi.log日志
4. 麻烦作者大大看一下啦~

**Bug Report Archive**
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/7878544/wifi.log)

zxystd, 2022-01-17:
日志不对，看 https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs

掉了重连之后是不是就不会再掉了？

oxygen-02, 2022-01-17:
> 日志不对，看 https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs
> 
> 掉了重连之后是不是就不会再掉了？

有时候也会掉，不清楚什么原因，我这两天多收集点日志反馈给你。这次的日志加载调试驱动给弄没了...

oxygen-02, 2022-01-18:
> 日志不对，看 https://openintelwireless.github.io/itlwm/Troubleshooting.html#runtime-logs
> 
> 掉了重连之后是不是就不会再掉了？

两次日常使用掉网，看看日志--->
 
[Log_2022-01-17_13-04-32-日常掉网第一次.log](https://github.com/OpenIntelWireless/itlwm/files/7885590/Log_2022-01-17_13-04-32-.log)

[Log_2022-01-18_09-39-21-日常第二次.log](https://github.com/OpenIntelWireless/itlwm/files/7885625/Log_2022-01-18_09-39-21-.log)
==========================================================
后续的日志也会继续上传上来，作者大大看看啦~

zxystd, 2022-01-18:
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7885686/AirportItlwm.kext.zip)
@oxygen-02 第一次掉线是固件崩溃，你试试这个会不会稳定些。第二次掉线是AP没反应，可能是2.4g信道阻塞了，建议连5g Wi-Fi。

oxygen-02, 2022-01-18:
> 

好的，我试试。
目前我使用的WiFi是 2.4G/5G 自动调整的。系统默认连接上的2.4G，可能不是很好测试5G上的稳定性

oxygen-02, 2022-01-18:
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/7885686/AirportItlwm.kext.zip) @oxygen-02 第一次掉线是固件崩溃，你试试这个会不会稳定些。第二次掉线是AP没反应，可能是2.4g信道阻塞了，建议连5g Wi-Fi。

AirportItlwm.kext v2.2.0 版本

Git拉取项目，拉取中掉WiFi。然后重新连接上WiFi。
日志：
[Log_2022-01-18_14-33-39.log](https://github.com/OpenIntelWireless/itlwm/files/7886799/Log_2022-01-18_14-33-39.log)

zxystd, 2022-01-18:
```
sending addba_req to 90:16:ba:72:45:40 on channel 10 mode 11n
itlwm: sending auth to 90:16:ba:72:44:d0 on channel 153 mode 11a
itlwm: trying AP 90:16:ba:72:45:50 on channel 64 instead
itlwm: sending auth to a8:49:4d:1a:78:a0 on channel 3 mode 11g
```
你这网络环境是真的奇葩，这么多信道合并ssid

oxygen-02, 2022-01-19:
> ```
> sending addba_req to 90:16:ba:72:45:40 on channel 10 mode 11n
> itlwm: sending auth to 90:16:ba:72:44:d0 on channel 153 mode 11a
> itlwm: trying AP 90:16:ba:72:45:50 on channel 64 instead
> itlwm: sending auth to a8:49:4d:1a:78:a0 on channel 3 mode 11g
> ```
> 
> 你这网络环境是真的奇葩，这么多信道合并ssid

emmmm，这个网络不是我自己的。大概率是因为把2.4G和5G合成一个WiFi导致的？
今天又是一样的状况。
Git拉取项目，拉取中掉WiFi。然后重新连接上WiFi。
[Log_2022-01-19_09-35-12.log](https://github.com/OpenIntelWireless/itlwm/files/7893451/Log_2022-01-19_09-35-12.log)

然后在回复你的过程中，又掉一次网，这是第二次掉网接近的日志
[Log_2022-01-19_09-39-33.log](https://github.com/OpenIntelWireless/itlwm/files/7893454/Log_2022-01-19_09-39-33.log)

其实在1.3版本我就在使用了，然后一直用到现在的2.2，使用的时候一般用的人多的WiFi貌似都有可能会掉....

oxygen-02, 2022-01-20:
日常掉连接
[Log_2022-01-20_09-39-27.log](https://github.com/OpenIntelWireless/itlwm/files/7901631/Log_2022-01-20_09-39-27.log)
[Log_2022-01-20_09-50-23.log](https://github.com/OpenIntelWireless/itlwm/files/7901663/Log_2022-01-20_09-50-23.log)

zxystd, 2022-01-20:
[image]
@oxygen-02 你这网络环境多个ssid一直在漫游，就是切换网络，这个没办法

closed 2022-03-25 by zxystd