OpenIntelWireless/itlwm #737: wifi repeatly connect and disconnect after wake up from sleep.

MHL3060, 2022-01-18:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 
- WiFi Card Model: intel 7260
- PCI Product ID: 
- macOS Version:  11.6.2

**Description**
the machine wakes up from a long sleep. wifi network connects and disconnects repeatly. 

```
[265420.716584]: itlwm: itlwm: sending addba_req to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265420.716599]: itlwm: iwm_tx_fill_cmd ridx=7
[265420.717517]: itlwm: taskq task_add ba_task add to queue
[265420.717527]: itlwm: iwm_ba_task start=1 ssn=0, tid=0 scd_queue=10
[265420.717535]: itlwm: iwm_enable_txq qid=10 tid=0 agg=1
[265420.717782]: itlwm: iwm_add_sta_cmd line=319
[265420.717904]: itlwm: iwm_sta_tx_agg tx agg done. err=0 status=1, mask_status=1
[265420.751997]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265420.767875]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265420.811258]: itlwm: itlwm: dumping device error log
[265420.811408]: itlwm: itlwm: Start Error Log Dump:
[265420.811409]: itlwm: itlwm: Status: 0x39, count: 6
[265420.811410]: itlwm: itlwm: 0x00001030 | ADVANCED_SYSASSERT          
[265420.811412]: itlwm: itlwm: 00800222 | trm_hw_status0
[265420.811413]: itlwm: itlwm: 00000000 | trm_hw_status1
[265420.811414]: itlwm: itlwm: 00000B30 | branchlink2
[265420.811415]: itlwm: itlwm: 000164C0 | interruptlink1
[265420.811421]: itlwm: itlwm: 00000000 | interruptlink2
[265420.811422]: itlwm: itlwm: DEADBEEF | data1
[265420.811423]: itlwm: itlwm: DEADBEEF | data2
[265420.811424]: itlwm: itlwm: DEADBEEF | data3
[265420.811425]: itlwm: itlwm: FFFB3777 | beacon time
[265420.811426]: itlwm: itlwm: F36B9855 | tsf low
[265420.811427]: itlwm: itlwm: 00000017 | tsf hi
[265420.811428]: itlwm: itlwm: 00000B2C | time gp1
[265420.811428]: itlwm: itlwm: 0042EE24 | time gp2
[265420.811429]: itlwm: itlwm: 00000000 | uCode revision type
[265420.811430]: itlwm: itlwm: 00000011 | uCode version major
[265420.811431]: itlwm: itlwm: BFB58538 | uCode version minor
[265420.811432]: itlwm: itlwm: 00000144 | hw version
[265420.811433]: itlwm: itlwm: 00089004 | board version
[265420.811434]: itlwm: itlwm: 0A02001C | hcmd
[265420.811435]: itlwm: itlwm: A6F23002 | isr0
[265420.811436]: itlwm: itlwm: 01C18000 | isr1
[265420.811437]: itlwm: itlwm: 00000112 | isr2
[265420.811438]: itlwm: itlwm: 00417CC8 | isr3
[265420.811439]: itlwm: itlwm: 00000000 | isr4
[265420.811439]: itlwm: itlwm: 10004112 | last cmd Id
[265420.811440]: itlwm: itlwm: 00000000 | wait_event
[265420.811441]: itlwm: itlwm: 000000C4 | l2p_control
[265420.811442]: itlwm: itlwm: 0001D410 | l2p_duration
[265420.811443]: itlwm: itlwm: 00000007 | l2p_mhvalid
[265420.811444]: itlwm: itlwm: 00000081 | l2p_addr_match
[265420.811445]: itlwm: itlwm: 00000015 | lmpm_pmg_sel
[265420.811446]: itlwm: itlwm: 15041745 | timestamp
[265420.811447]: itlwm: itlwm: 0000E8F0 | flow_handler
[265420.811448]: itlwm: driver status:
[265420.811449]: itlwm:   tx ring  0: qid=0  cur=59  queued=0  
[265420.811450]: itlwm:   tx ring  1: qid=1  cur=0   queued=0  
[265420.811452]: itlwm:   tx ring  2: qid=2  cur=0   queued=0  
[265420.811453]: itlwm:   tx ring  3: qid=3  cur=0   queued=0  
[265420.811454]: itlwm:   tx ring  4: qid=4  cur=0   queued=0  
[265420.811468]: itlwm:   tx ring  5: qid=5  cur=95  queued=0  
[... 212 lines cut ...]
[265424.433502]: itlwm: ieee80211_send_eapol_key
[265424.433503]: itlwm: ieee80211_eapol_key_mic
[265424.433544]: itlwm: ieee80211_must_update_group_key
[265424.433545]: itlwm: ieee80211_recv_4way_msg3 k_len=16 rsc=7B 29 04 00 00 00itlwm: ieee80211_set_key kid=1 cipher=8, error=0
[265424.433553]: itlwm: ieee80211_set_link_state nstate=4, old_state=2
[265424.433554]: itlwm: ieee80211_set_link_state LINK_STATE_IS_UP
[265424.435515]: Reinit'd ND information for interface en1
[265424.435556]: en1: starting optimistic DAD immediately for fe80:5::1c52:cf53:4ce:3e7e
[265424.435561]: itlwm: setMulticastList multicast count=5 bssid=9c:1e:95:a7:12:46
[265424.438227]: itlwm: taskq task_add ba_task add to queue
[265424.438306]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265424.438317]: itlwm: iwm_tx_fill_cmd ridx=7
[265424.440302]: itlwm: setMulticastList multicast count=6 bssid=9c:1e:95:a7:12:46
[265424.453263]: itlwm: taskq task_add ba_task add to queue
[265424.453337]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265424.453349]: itlwm: iwm_tx_fill_cmd ridx=7
[265424.467951]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265424.479781]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265424.837912]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265424.863842]: USB Sound assertion (Streams wrapping on different frames (off by more than 2)) in /System/Volumes/Data/SWE/macOS/BuildRoots/9dc5c14c79/Library/Caches/com.apple.xbs/Sources/AppleUSBAudio/AppleUSBAudio-405.39/KEXT/AppleUSBAudioStream.cpp at line 3419
[265424.909845]: itlwm: taskq task_add ba_task add to queue
[265424.910331]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 136 mode 11ac
[265424.910342]: itlwm: iwm_tx_fill_cmd ridx=7
[265425.013812]: tx_flush:1182: disk3 xid 5275918 tx stats: # 201420 finish 201535 enter 95 wait 1376 1862us close 2us flush 1901us
```

zxystd, 2022-01-23:
Can you try the latest alpha version to see if it is more better?

MHL3060, 2022-01-23:
Yes. the issue persists. No improvement on the latest snapshot version.

MHL3060, 2022-01-23:
oops. there is a new snapshot version. let me try that. (I was trying the one 5days ago).

zxystd, 2022-01-25:
Does your router enabled SMPS? Have you tried another router like the cellphone hotspot?

MHL3060, 2022-01-26:
My router seems don't have the SMPS option. 
In the last two days, I have been using this snapshot itlwm-v2.2.0-DEBUG-alpha-09a02e0 and it seems to have improved but still seeing the issue intermittently. I will give it a try for a longer time and report what I found.

MHL3060, 2022-01-28:
Switching to a different router doesn't help. here is more info that I observed. in the network preference, once the wifi card cannot get the ip address, put the machine to sleep and wake it up after the sleep, sometimes after a few connect/disconnects, it becomes stable and able to get an IP address.

MHL3060, 2022-02-10:
so. the issue is intermittent. after wakeup, some time it can get a stable connection after a few connect/disconnect.
sometime it give up retry after a few connect disconnect.

Yes. I have tried to turn on/off the wifi using heliport. 
the strange thing is sometime heliport is able to turn off the wifi. sometime it cannot. 

in terms of connectivity. After I toggle the on/off. it start to do the connect/disconnect again. sometime it can get into a stable connectivity. sometime it does. 

it is like lottery.

MHL3060, 2022-02-17:
one thing to note is the latest snapshot `itlwm-v2.2.0-DEBUG-alpha-6050e96.zip` seems improve a bit. It seems have a better chance to get reconnect after many cycles of connect/disconnect. I will test it more.

MHL3060, 2022-02-21:
I think it seems not able to connects to 11AC mode. if it tries to connect 11AC mode. the firmware crash 
```
 itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:46 on channel 112 mode 11ac
[13071.572784]: itlwm: iwm_tx_fill_cmd ridx=7
[13071.648657]: Sandbox: 1 duplicate report for icdd allow network-outbound /private/var/run/mDNSResponderSandbox: rapportd(378) deny(1) mach-lookup com.apple.lsd.mapdbSandbox: 1 duplicate report for rapportd deny(1) mach-lookup com.apple.lsd.mapdbSandbox: icdd(3225) allow network-outbound /private/var/run/mDNSResponderSandbox: 1 duplicate report for icdd allow network-outbound /private/var/run/mDNSResponderSandbox: icdd(3225) allow file-read-data /Library/Image Capture/DevicesSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner3.app/Contents/Resources/DeviceMatchingInfo.plistSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner2.app/Contents/Resources/DeviceMatchingInfo.plistSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner1.app/Contents/Resources/DeviceMatchingInfo.plistSandbox: icdd(3225) allow file-read-data /Library/Image Capture/DevicesSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner3.app/Contents/Resources/DeviceMatchingInfo.plistSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner2.app/Contents/Resources/DeviceMatchingInfo.plistSandbox: icdd(3225) allow file-read-data /Library/Image Capture/Devices/Canon MFScanner1.app/Contents/Resources/DeviceMatchingInfo.plistitlwm: itlwm: sending addba_req to 9c:1e:95:a7:12:46 on channel 112 mode 11ac
[13072.024672]: en1: DAD complete for fe80:5::4e1:c25a:54ba:9ab4 - no duplicates found .
[13072.024721]: itlwm: iwm_tx_fill_cmd ridx=7
[13072.025898]: itlwm: taskq task_add ba_task add to queue
[13072.025909]: itlwm: iwm_ba_task start=1 ssn=0, tid=0 scd_queue=10
[13072.025918]: itlwm: iwm_enable_txq qid=10 tid=0 agg=1
[13072.025953]: itlwm: iwm_add_sta_cmd line=319
[13072.026402]: itlwm: iwm_sta_tx_agg tx agg done. err=0 status=1, mask_status=1
[13072.033919]: itlwm: itlwm: dumping device error log
[13072.034023]: itlwm: itlwm: Start Error Log Dump:
[13072.034024]: itlwm: itlwm: Status: 0x39, count: 6
[13072.034026]: itlwm: itlwm: 0x00001030 | ADVANCED_SYSASSERT
[13072.034027]: itlwm: itlwm: 00800223 | trm_hw_status0
[13072.034028]: itlwm: itlwm: 00000000 | trm_hw_status1
[13072.034029]: itlwm: itlwm: 00000B30 | branchlink2
[13072.034030]: itlwm: itlwm: 000164C0 | interruptlink1
[13072.034031]: itlwm: itlwm: 00000000 | interruptlink2
[13072.034032]: itlwm: itlwm: DEADBEEF | data1
[13072.034033]: itlwm: itlwm: DEADBEEF | data2
[13072.034034]: itlwm: itlwm: DEADBEEF | data3
[13072.034035]: itlwm: itlwm: FFF9FA0B | beacon time
[13072.034036]: itlwm: itlwm: FD4E45F0 | tsf low
[13072.034037]: itlwm: itlwm: 0000001D | tsf hi
[13072.034038]: itlwm: itlwm: 00000BAA | time gp1
[13072.034039]: itlwm: itlwm: 00420B0A | time gp2
[13072.034040]: itlwm: itlwm: 00000000 | uCode revision type
[13072.034041]: itlwm: itlwm: 00000011 | uCode version major
[13072.034042]: itlwm: itlwm: BFB58538 | uCode version minor
[13072.034043]: itlwm: itlwm: 00000144 | hw version
[13072.034044]: itlwm: itlwm: 00089004 | board version
[13072.034045]: itlwm: itlwm: 0A01001C | hcmd
[13072.034046]: itlwm: itlwm: 24023082 | isr0
[13072.034046]: itlwm: itlwm: 00818000 | isr1
[13072.034047]: itlwm: itlwm: 00000012 | isr2
[13072.034048]: itlwm: itlwm: 004128C0 | isr3
[13072.034049]: itlwm: itlwm: 00000000 | isr4
[13072.034050]: itlwm: itlwm: 01004112 | last cmd Id
[13072.034051]: itlwm: itlwm: 00000000 | wait_event
[13072.034052]: itlwm: itlwm: 000000D4 | l2p_control
[13072.034053]: itlwm: itlwm: 0001DC10 | l2p_duration
[13072.034054]: itlwm: itlwm: 00000007 | l2p_mhvalid
[13072.034054]: itlwm: itlwm: 00000081 | l2p_addr_match
[13072.034055]: itlwm: itlwm: 00000015 | lmpm_pmg_sel
[13072.034056]: itlwm: itlwm: 15041745 | timestamp
[13072.034057]: itlwm: itlwm: 00007078 | flow_handler
[13072.034058]: itlwm: driver status:
[13072.034059]: itlwm:   tx ring  0: qid=0  cur=64  queued=1
[13072.034061]: itlwm:   tx ring  1: qid=1  cur=0   queued=0
[13072.034062]: itlwm:   tx ring  2: qid=2  cur=0   queued=0
[13072.034063]: itlwm:   tx ring  3: qid=3  cur=0   queued=0
[13072.034065]: itlwm:   tx ring  4: qid=4  cur=0   queued=0
[13072.034066]: itlwm:   tx ring  5: qid=5  cur=208 queued=0
[13072.034067]: itlwm:   tx ring  6: qid=6  cur=0   queued=0
[13072.034068]: itlwm:   tx ring  7: qid=7  cur=0   queued=0
[13072.034069]: itlwm:   tx ring  8: qid=8  cur=0   queued=0
[13072.034071]: itlwm:   tx ring  9: qid=9  cur=0   queued=0
[13072.034072]: itlwm:   tx ring 10: qid=10 cur=3   queued=2
[13072.034073]: itlwm:   tx ring 11: qid=11 cur=0   queued=0
[13072.034074]: itlwm:   tx ring 12: qid=12 cur=0   queued=0
[13072.034075]: itlwm:   tx ring 13: qid=13 cur=0   queued=0
[13072.034076]: itlwm:   tx ring 14: qid=14 cur=0   queued=0
[13072.034078]: itlwm:   tx ring 15: qid=15 cur=0   queued=0
[13072.034079]: itlwm:   tx ring 16: qid=16 cur=0   queued=0
[13072.034080]: itlwm:   tx ring 17: qid=17 cur=0   queued=0
[13072.034081]: itlwm:   tx ring 18: qid=18 cur=0   queued=0
[13072.034082]: itlwm:   tx ring 19: qid=19 cur=0   queued=0
[... 24 lines cut ...]
[13076.550779]: itlwm: iwm_ba_task start=1 ssn=0, tid=0 scd_queue=10
[13076.550788]: itlwm: iwm_enable_txq qid=10 tid=0 agg=1
[... cut ...]
[13080.561111]: itlwm: itlwm: sending addba_resp to 9c:1e:95:a7:12:42 on channel 11 mode 11n
[13080.561124]: itlwm: iwm_tx_fill_cmd ridx=8
[13080.674636]: en1: starting optimistic DAD immediately for 2001:569:7cd7:4a00:4d6:e76e:d31d:f697
[13080.674665]: itlwm: setMulticastList multicast count=7 bssid=9c:1e:95:a7:12:42
[13080.674677]: en1: starting optimistic DAD immediately for 2001:569:7cd7:4a00:5cc1:c404:54c6:d6a8
[13080.674686]: itlwm: setMulticastList multicast count=8 bssid=9c:1e:95:a7:12:42
````

zxystd, 2022-02-21:
I had reproduce this issue with ac9260 + hotspot with Redmi K20P, when issue happened, the driver connect to the WiFi and then firmware crashed, but it can connect to other APs with no issue. Also I can't connect this AP with Windows and Linux, so I am sure that it is a issue from the AP or the Intel firmware inside, we can't do any fix on that(the code is based on Linux but Linux behaves the same).

MHL3060, 2022-02-21:
my experience is a bit different than yours. When I boot in the windows 10 or Ubuntu  20.04. I can able to connect to the wifi without issue. (my car is ac7260 however)

MHL3060, 2022-02-21:
the latest version of the itlwm seems improved a bit. when cold boot, it is able to connect to the wifi with a short period of retries.

MHL3060, 2022-03-05:
I realized the connect/disconnect symptom comes from trying to connect to `A` or `AC` or `N` channel > 12.
when connecting only to N mode only router, I don't see the connect/disconnect so far.

my question is can we introduce a boot-args flag to dictate what mode it can connect to?

KIKOcaoyue, 2022-03-27:
> Does your router enabled SMPS? Have you tried another router like the cellphone hotspot?

I met the same problem, I often work under different router environments, and some routers doesn't cause the problem, while some routers cause (including cellphone hotspot).

zxystd, 2023-06-11:
> I realized the connect/disconnect symptom comes from trying to connect to `A` or `AC` or `N` channel > 12. when connecting only to N mode only router, I don't see the connect/disconnect so far.
> 
> my question is can we introduce a boot-args flag to dictate what mode it can connect to?

`-novht`

closed 2023-06-11 by zxystd