OpenIntelWireless/itlwm #738: Kernel panic in ieee80211_ra_get_rateset

pigworlds, 2022-01-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: 9560
- PCI Product ID: 8086 A370
- macOS Version: 21C52

**Description**
This driver is very stable. This is the first time I hit KP in a few months. The mcs=12 seems odd.

```
panic(cpu 0 caller 0xffffff8019b281c8): ieee80211_ra_get_rateset mcs=12 rs_count=16 sgi=0 nss=2 bw=5 rate==NULL!!!!

Panicked task 0xffffff8bd134e670: 223 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff956a681aa0, Frame : Return Address
0xffffffd110deb650 : 0xffffff8015087ddd mach_kernel : _handle_debugger_trap + 0x41d
0xffffffd110deb6a0 : 0xffffff80151e7d15 mach_kernel : _kdp_i386_trap + 0x145
0xffffffd110deb6e0 : 0xffffff80151d74e3 mach_kernel : _kernel_trap + 0x533
0xffffffd110deb730 : 0xffffff8015027a60 mach_kernel : _return_from_trap + 0xe0
0xffffffd110deb750 : 0xffffff80150881ad mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffd110deb870 : 0xffffff8015087966 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffd110deb8d0 : 0xffffff80159160b9 mach_kernel : _panic + 0x54
0xffffffd110deb940 : 0xffffff8019b281c8 com.zxystd.AirportItlwm : __Z24ieee80211_ra_get_ratesetP17ieee80211_ra_nodeP12ieee80211comP14ieee80211_nodei + 0x1d8
0xffffffd110deb9b0 : 0xffffff8019bcc5e2 com.zxystd.AirportItlwm : __ZN6ItlIwm26iwm_ht_single_rate_controlEP9iwm_softcP14ieee80211_nodehhhi + 0x82
0xffffffd110deba30 : 0xffffff8019bcd9e9 com.zxystd.AirportItlwm : __ZN6ItlIwm17iwm_ampdu_tx_doneEP9iwm_softcP14iwm_cmd_headerP8iwm_nodeP11iwm_tx_ringjhhtiP17iwm_agg_tx_status + 0x439
0xffffffd110debb20 : 0xffffff8019bce34d com.zxystd.AirportItlwm : __ZN6ItlIwm13iwm_rx_tx_cmdEP9iwm_softcP13iwm_rx_packetP11iwm_rx_data + 0x3cd
0xffffffd110debc00 : 0xffffff8019bad0a9 com.zxystd.AirportItlwm : __ZN6ItlIwm10iwm_rx_pktEP9iwm_softcP11iwm_rx_dataP9mbuf_list + 0x8a9
0xffffffd110debe00 : 0xffffff8019bd581e com.zxystd.AirportItlwm : __ZN6ItlIwm14iwm_notif_intrEP9iwm_softc + 0xde
0xffffffd110debe60 : 0xffffff8019bd5e84 com.zxystd.AirportItlwm : __ZN6ItlIwm8iwm_intrEP8OSObjectP22IOInterruptEventSourcei + 0x5c4
0xffffffd110debed0 : 0xffffff8015858d13 mach_kernel : __ZN22IOInterruptEventSource12checkForWorkEv + 0x163
0xffffffd110debf20 : 0xffffff801585756e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x13e
0xffffffd110debf60 : 0xffffff8015856b97 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x37
0xffffffd110debfa0 : 0xffffff801502718e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.1)[FA02D059-D593-3D8E-A031-1F99EA4D67EB]@0xffffff8019b27000->0xffffff801aac0fff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[D73065D2-A7FE-35CE-A8C4-E0FBFFCCED6B]@0xffffff8016ded000->0xffffff8016f33fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[39332FD1-54B0-3F6A-9FF1-2233888F6C5A]@0xffffff801794c000->0xffffff8017962fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[023CD3FF-4EFC-3455-89C2-FF8BD98013EB]@0xffffff8017be6000->0xffffff8017c10fff

Process name corresponding to current thread (0xffffff956a681aa0): kernel_task
Boot args: -v keepsyms=1 -hbfxbeta chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21C52

Kernel version:
Darwin Kernel Version 21.2.0: Sun Nov 28 20:28:54 PST 2021; root:xnu-8019.61.5~1/RELEASE_X86_64
Kernel UUID: E32C293C-7D8B-327C-A335-6AAD250A4881
KernelCache slide: 0x0000000014e00000
KernelCache base:  0xffffff8015000000
Kernel slide:      0x0000000014e10000
Kernel text base:  0xffffff8015010000
__HIB  text base: 0xffffff8014f00000
System model name: MacBookPro15,1 (Mac-937A206F2EE63C01)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 12494622269460
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000b5d2185a403
  Sleep   : 0x00000b37bdec01b3 0x0000000111dc6fec 0x00000b2c6c0c1d68
  Wake    : 0x00000b37c55e8fcb 0x0000000112ca8ade 0x00000b37bfb4ba83
Zone info:
Foreign   : 0xffffff801fe7c000 - 0xffffff801fe89000
Native    : 0xffffff809e14e000 - 0xffffffa09e14e000
Readonly  : 0xffffff856ae1a000 - 0xffffff87047ae000
Metadata  : 0xffffffdb4721e000 - 0xffffffdb67a01000
Bitmaps   : 0xffffffdb67a01000 - 0xffffffdb7fa01000
```

**Bug Report Archive**

**Kext Download Source**
v2.1.0 stable

zxystd, 2022-01-20:
Thank you for the feedback!
Seems that the Tx report is invalid and we should ignore it instead of trigger kernel panic, isn't it?

pigworlds, 2022-01-20:
The code base is quite solid, and I've been using it for a couple of months without any panic. I am not sure the invalid value causes the panic is coming from the firmware or the driver.
I am worry that ignore the value may hide other stability bugs. Maybe we can leave this issue for a while and waiting for more reproduce.

zxystd, 2022-01-21:
It should be ignore, if reported mcs isn't match the mcs we are expected(send in LQ command).
The diff is below
```
diff --git a/itlwm/hal_iwm/mac80211.cpp b/itlwm/hal_iwm/mac80211.cpp
index 7bba7f0..a5f414d 100644
--- a/itlwm/hal_iwm/mac80211.cpp
+++ b/itlwm/hal_iwm/mac80211.cpp
@@ -1067,8 +1067,7 @@ iwm_ht_single_rate_control(struct iwm_softc *sc, struct ieee80211_node *ni,
     struct ieee80211com *ic = (struct ieee80211com *)&sc->sc_ic;
     struct iwm_node *wn = (struct iwm_node *)ni;
     int mcs = rate;
-    const struct ieee80211_ra_rate *rs =
-    ieee80211_ra_get_rateset(&wn->in_rn, ic, ni, rate);
+    const struct ieee80211_ra_rate *rs;
     unsigned int retries = 0, i;
 
     /*
@@ -1085,6 +1084,7 @@ iwm_ht_single_rate_control(struct iwm_softc *sc, struct ieee80211_node *ni,
 
     wn->lq_rate_mismatch = 0;
 
+    rs = ieee80211_ra_get_rateset(&wn->in_rn, ic, ni, rate);
     /*
      * Firmware has attempted rates in this rate set in sequence.
      * Retries at a basic rate are counted against the minimum MCS.
diff --git a/itlwm/hal_iwn/ItlIwn.cpp b/itlwm/hal_iwn/ItlIwn.cpp
index d9f8b7c..99d6554 100644
--- a/itlwm/hal_iwn/ItlIwn.cpp
+++ b/itlwm/hal_iwn/ItlIwn.cpp
@@ -2377,8 +2377,7 @@ iwn_ht_single_rate_control(struct iwn_softc *sc, struct ieee80211_node *ni,
     struct ieee80211com *ic = &sc->sc_ic;
     struct iwn_node *wn = (struct iwn_node *)ni;
     int mcs = rate;
-    const struct ieee80211_ra_rate *rs =
-    ieee80211_ra_get_rateset(&wn->rn, ic, ni, rate);
+    const struct ieee80211_ra_rate *rs;
     unsigned int retries = 0, i;
     
     /*
@@ -2395,6 +2394,7 @@ iwn_ht_single_rate_control(struct iwn_softc *sc, struct ieee80211_node *ni,
     
     wn->lq_rate_mismatch = 0;
     
+    rs = ieee80211_ra_get_rateset(&wn->rn, ic, ni, rate);
     /*
      * Firmware has attempted rates in this rate set in sequence.
      * Retries at a basic rate are counted against the minimum MCS.
```

pigworlds, 2022-01-21:
This looks great!

closed 2022-01-24 by pigworlds