OpenIntelWireless/itlwm #740: 使用AirportItlwm后，电脑偶发性的死机，只能重启

Venus-Chen, 2022-01-26:
如题，使用AirportItlwm后，电脑偶发性的死机，只能重启。
从1.2版本的一直升级到现如今的2.1版本，都存在这个问题，重启后，系统会弹出错误报告，打开来看了之后，里面有AirportItlwm相关的信息，具体错误内容如下：
[未命名.txt](https://github.com/OpenIntelWireless/itlwm/files/7938595/default.txt)

williambj1, 2022-02-09:
@zxystd

Venus-Chen, 2022-02-10:
My Computer's model is Lenovo Y9000X 2020, here is its detail:
Mother Board：        HM370
CPU：                        i7-9750H
Screen Resolution：1080P
SSD：                        SN750 1T
NIC：                         Intel AX200

zxystd, 2022-02-11:
Same issue with #596

Venus-Chen, 2022-02-11:
This question is quite similar with #596 , but it happened in a different situation. my system's version is 11.6.2 and i use v0.7.5 of OpenCore.
I use the latest stable version(v2.1), i opened my computer at about 8 am, about 2 or 3 hours later, my system crashed, it becomes irresponsible, i had to stop it by press the power button for 10s, then reboot it, the system popup a window, then i got this error message.
But this question is very difficult to reproduce, sometimes it occurred 2 or 3 times a day, sometimes it never occur for 1 or 2 weeks or even more.

zxystd, 2022-03-25:
can you check if the v2.2.0-alpha have the same panic?

Venus-Chen, 2022-03-25:
ok, once i back to my office, i'll check this version immediately.

Venus-Chen, 2022-07-22:
今年3月份，上海疫情，个人被封闭在家居家办公3个半月，笔记本则是一直放在公司里面，一直到7月4号正式上班后，才继续使用笔记本。截止到目前已经使用了3周了，还没有出现死机的情况，但是WI-FI还是依旧会间接性的断连然后重连。我用的Itlwm的版本还是2.1.0稳定版

einsqing, 2023-08-16:
> 今年3月份，上海疫情，个人被封闭在家居家办公3个半月，笔记本则是一直放在公司里面，一直到7月4号正式上班后，才继续使用笔记本。截止到目前已经使用了3周了，还没有出现死机的情况，但是WI-FI还是依旧会间接性的断连然后重连。我用的Itlwm的版本还是2.1.0稳定版

同样的问题，如何解决？

closed 2024-03-16 by zxystd