OpenIntelWireless/itlwm #741: Intel AX200 often disconnects and reconnects

stefan89, 2022-01-30:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: itlwm 2.0.0
- WiFi Card Model: Intel AX200
- PCI Product ID: Asus PCE-AX3000
- macOS Version: 11.6.2

**Description**
Currently I have the issue that my WIFI disconnects often and after a few seconds it reconnects. This is especially very annoying when doing a video call or a downloading large file. 

The following bug report is created shortly after I faced a disconnect / reconnect.

**Bug Report Archive**
[bugreport_62113.zip](https://github.com/OpenIntelWireless/itlwm/files/7965964/bugreport_62113.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v2.0.0

stefan89, 2022-01-31:
FYI I also tried v2.2.0-alpha, v2.1.0, v1.3.0, v1.20 of itwlm.kext. For all these versions I am facing the same issue.

stefan89, 2022-01-31:
Update: it seems that I don't have issues when I use my phone as hotspot.

What could be the reason for this?

fortoszone, 2022-02-03:
can confirm here. also have an ax200 and wifi just disconnected by itself sometimes

kphanipavan, 2022-02-07:
I can +1 on this.

I have anl Killer Ax1650i, which is essentially an Intel AX201.

I also want to report that wifi disconnects happen when the system is trying to scan for other nerworks.

I was able to connect, close the wifi applet and successfully download MacOS from recovery. Disconnects happening only when it scans for other networks.

williambj1, 2022-02-09:
@stefan89 The firmware crashed in your case when the driver reasssociates with the network, will need more help from @zxystd.

@kphanipavan @fortoszone Logs are needed to help identify the issue.

KIKOcaoyue, 2022-03-27:
> 

Is there a switch so that the computer does not automatically search for other networks?

zxystd, 2022-03-28:
> Is there a switch so that the computer does not automatically search for other networks?

what's the meaning of 'automatically search for other networks'?

zxystd, 2022-04-03:
@stefan89 Please try the latest alpha version.

Creepo513, 2022-07-26:
Also posting here that using my phone as a Hotspot has 0 issues while my home network often disconnects or thinks it's connected but has no internet. Tried every version that supports Monterey, no luck

HEPBO3AH, 2023-01-07:
Started happening to me today. Was working fine for a long time. Didn't upgrade any kexts or anything.

closed 2024-03-16 by zxystd