OpenIntelWireless/itlwm #742: Wifi shows it is connected but doesn't work.

Fuzail-maker, 2022-01-31:
Hi, I don't like to shutdown my laptop so I usually just close the lid to put it to sleep. I don't really know how to reproduce this, it happens randomly that the wifi toggle shows I'm connected to the wifi but if I try to load something like google.com, it doesn't load. Toggling off the wifi, tho the wifi toggle switches to off position , the wifi remains connected and the 3 signal bars stay there. Only a reboot fixes this!

I do not have any logs right now as it happens randomly, I'll try to grab them next time. But meanwhile, [here's](https://imgur.com/a/fY9rZTN) the wifi card that came with my laptop. And [this](https://drive.google.com/drive/folders/1p5BzVDh9_jvmC9zerKlp6BRRImyMvcu3?usp=sharing) is my EFI.

stefan89, 2022-01-31:
Maybe a silly question, but is the card correctly wired? See red circle:
[image]

stefan89, 2022-01-31:
PS. have you tried itwlm.kext instead of AirportItlwm.kext? If I understand it correctly, it should be more stable

Fuzail-maker, 2022-01-31:
@stefan89 yup, the black antennae cable came out when I took the card out for taking a picture, I did put it back before installing it. Nope I haven't tried just the itwlm, ill have a look. Thanks

stefan89, 2022-01-31:
If you're trying itwlm.kext, don't forget to install heliport app first: https://github.com/OpenIntelWireless/HeliPort

Fuzail-maker, 2022-02-01:
@stefan89 ohh tysm, I'll try it tomorrow and update here

Fuzail-maker, 2022-02-01:
@stefan [rip](https://i.imgur.com/qdIREfp.jpg) I am a student and I use the my university's wifi all the time. Unfortunately heliport doesn't support it :/

Works with airportitlwm tho.

williambj1, 2022-02-09:
Please upload logs.

closed 2022-02-09 by williambj1