OpenIntelWireless/itlwm #743: Display Artifacts with WiFi On

Shotz5, 2022-02-01:
Hey guys,

I'm having a very weird issue on my Hackintosh running macOS Big Sur 11.6.3 with an Intel AX200 and UHD 630 Graphics... whenever I turn on WiFi and am actively using the WiFi, the display will start periodically flickering. The more I use the WiFi, the worse the issue gets. The issue not present when using a USB to Ethernet adapter. Clean install does not fix. I believe it may be an issue with the plugin and my configuration, but I can't pinpoint what. Consulting Google provided no information as well...

Debug File: [bugreport_3670.zip](https://github.com/OpenIntelWireless/itlwm/files/7975608/bugreport_3670.zip)

config.plist: [config.plist.zip](https://github.com/OpenIntelWireless/itlwm/files/7975616/config.plist.zip)

Thanks for any help or insight anyone can give?
Shotz

Shotz5, 2022-02-07:
So I have found out that it was actually a GFX driver related issue, that just occured significantly more when using WiFi. No issue here.

Thanks

closed 2022-02-07 by Shotz5

phu54321, 2022-02-14:
Hello, I'm having a same issue today. Could you elaborate what you've done? Thanks.

My display looks like this: https://www.reddit.com/r/hackintosh/comments/ssaxug/photosensitive_epilepsy_alert_screen_moving/

Shotz5, 2022-02-15:
Yes that does appear to be similar to the issue I was experiencing, the way to confirm would be to go to Console and look for kernel logs saying "display underflow" or something along those lines... if you are experiencing those issues, see link below for the DeviceProperties I used to fix it.

https://github.com/Shotz5/Razer-Blade-Hackintosh#graphics-1

phu54321, 2022-02-15:
Wow your patch fixed the issue. Thanks!