OpenIntelWireless/itlwm #744: 802.1x doesn't work in 2.2.0-alpha

terminet85, 2022-02-04:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version:  v2.2.0-alpha
- WiFi Card Model: 
- PCI Product ID: 
- macOS Version: 12.2

**Description**
802.1.x and iPhone paring doesn't work on 2.2.0-alpha. In 2.2.1 works but mostly buggy

williambj1, 2022-02-09:
@terminet85 

> iPhone paring

Not sure what this means, could you please explain a bit more?

> In 2.2.1 works

AFAIK this version does not exist. Is it from another source?

Logs are mandatory to help identify the issue.

terminet85, 2022-02-09:
> Not sure what this means, could you please explain a bit more?

I’m talking about Instant Hotspot 

> AFAIK this version does not exist. Is it from another source?

2.1.0 mistype…
 
> Logs are mandatory to help identify the issue.

no problem. Any hints or manual how to collect?

williambj1, 2022-02-09:
> I’m talking about Instant Hotspot

Not supposed to work right now, you'll need to manually turn it on from your iPhone.

> no problem. Any hints or manual how to collect?

Can be found in the docs.

williambj1, 2022-03-03:
@terminet85 Hi, any progress?

terminet85, 2022-03-03:
Sorry I’m busy with my work… I’m closing this and make new one when I can.

closed 2022-03-03 by terminet85

williambj1, 2022-03-03:
@terminet85  Just for ref, I've managed to successfully connect to a local library's Wi-Fi that uses 802.1x WPA2-Enterprise for authentication using AirportItlwm without facing any issues. Take your time and have a nice day.